alphaville/optimization-engine #7: Set initial guess over UDP

alphaville, 2019-03-24:
**Is your feature request related to a problem? Please describe.**
The user may need to update (or reset) the initial guess over UDP.

**Describe the solution you'd like**
The issue is that if an instance of the optimizer fails to converge, the approximate solution will be provided to the next instance. In some cases, the solver may return `NaN` as the solution. Such solutions should never be used to initialise the next instance of the solver.

**Describe alternatives you've considered**

**Additional context**
Once implemented, the documentation has to be updated.

alphaville, 2019-06-02:
Done in Python

korken89, 2019-08-18:
I think we can close this?

closed 2020-05-12 by alphaville