alphaville/optimization-engine #8: C/C++ interface

mvukov, 2019-03-25:
Hello,

Are there any plans to make a C and/or C++ (my personal preference) interface?

Cheers!

alphaville, 2019-03-25:
@mvukov Hi Milan. You can interface Optimization Engine from C/C++ using the UDP interface. Would that be OK for you?

Building a native C++ interface shouldn't be too hard. I will try to address that.

mvukov, 2019-03-25:
Hi Pantelis,

It would be nice to have a native C++ interface -- simpler to use IMHO. Nothing urgent on my end, please ping me once you guys have something. Thanks in advance :)

Cheers,
Milan

sagniknitr, 2019-03-29:
@alphaville I am interested to work in this...let me how are you planning to design the interface for C/C++..

alphaville, 2019-04-02:
Hi Sagnik @sagniknitr. We will be moving on with this issue very soon. My colleague Emil Fresk pointed out to [this documentation](https://rust-embedded.github.io/book/interoperability/rust-with-c.html). Apparently it's easy to build a C interface for Rust code. This can also be called from C++.

The way OpEn works right now, is that the user [generates Rust code automatically in MATLAB](https://alphaville.github.io/optimization-engine/docs/matlab-interface). The generated code has a main function (`main.rs`) which exports its functionality over a UDP socket. We need to redesign this a little.

The idea is for the auto-generated code to implement the following three methods in Rust:

- an *init* method, which will be called once to initialize the parametric solver (create necessary objects)
- a *call* method to which the user will provide a parameter `p` and obtain an optimal solution `u` along with solver statistics (number of iterations, solution time, etc)
- potentially, a *destroy* method

These three methods will be exported to C/C++.

Then, the auto-generated code will involve a second layer, written in Rust, that will export the said functionality over UDP sockets.

How would you like to be involved? You are welcome to join our discussions on [Discord](https://discordapp.com/invite/mfYpn4V). We can arrange a meeting to discuss how to move on with it.

korken89, 2019-05-15:
Hi, I'd like to make people aware that I'm currently implementing the C/C++ support :)

korken89, 2019-06-04:
The experimental interface is now available in `open-clib`, this will now be transferred to the Python code generator so any solver will automatically come with C/C++ bindings.

closed 2019-06-07 by alphaville

alphaville, 2019-06-07:
@mvukov @sagniknitr @korken89 OpEn now supports the generation of a C/C++ interface (header files) and static and dynamic files you can use. We will update the documentation soon. If you would like to try this very quickly, download the source code from `master`, install Rust and opengen as follows:

- Install Rust following [these instructions](https://www.rust-lang.org/tools/install)
- Install Opegen as follows:

```
cd opengen
pip install virtualenv
virtualenv venv
source venv/bin/activate
python setup.py install
```

- Install CasADi:

```python
pip install casadi
```

- try out the following python code:

```python
import opengen as og
import casadi.casadi as cs

# Build parametric optimizer
# ------------------------------------
u = cs.SX.sym("u", 5)
p = cs.SX.sym("p", 2)
phi = og.functions.rosenbrock(u, p)
problem = og.builder.Problem(u, p, phi)
build_config = og.config.BuildConfiguration() \
    .with_build_directory("autogenerated_optimizers") \
    .with_build_mode("release")\
    .with_rebuild(True)
meta = og.config.OptimizerMeta() \
    .with_optimizer_name("my_optimizer")
builder = og.builder.OpEnOptimizerBuilder(problem, meta,
                                          build_config) \
    .with_verbosity_level(1)

# Generate C bindings
builder.enable_c_bindings_generation()
builder.build()
```

This will build a parametric optimizer (in Rust) and will store all auto-generated files in:

```
./autogenerated_optimizers/my_optimizer/
```

Relative to where you run the code from (you can use absolute paths if you prefer).

Once the code runs, you should be able to find two header files therein: `open_bindings.h` and `open_bindings.hpp`. 

You will find the static and dynamic library files in 

```
./autogenerated_optimizers/my_optimizer/target/release/
```

Namely, `libmy_optimizer.a` and `libmyoptimizer.so`.

- to use this optimizer in `C` (similarly, in `C++`), create a main file like this one:

```c
// File: optimizer.c 

#include <stdio.h>
#include "open_bindings.h"

int main() {

	HomotopyCache * y = solver_new();
	const double p[2] = {1.0, 10.0};
	double u[5] = {0};
	CHomotopySolverStatus status = solver_solve(y, u, p);

	int i;
	for (i=0; i<5; ++i) {
		printf("u[%d] = %g\n", i, u[i]);
	}

	printf("exit status = %d\n", status.exit_status);
	printf("iterations = %lu\n", status.num_inner_iterations);
	printf("outer iterations = %lu\n", status.num_outer_iterations);

	solver_free(y);

	return 0;	
}
```

and compile it with

```
gcc  optimizer.c -l:libmy_optimizer.a -L./target/release -pthread -lm -ldl -o optimizer
```

PR #15 (2019-06-03): [do not merge yet] C/C++ library export
Hi Pantelis,

I have a working example of generating a static C/C++ library based on PANOC now.
But I found an issue we have to tackle. Currently the generation of bounds allocate as we need to pass `Vec<_>`s into the constraints.
I think we should change that to `&[_]`, or have an API to move the vectors out again when we need to destroy the `Problem` struct. 

I was hoping to be able to have the `Problem` struct as a part of the `PanocInstance`, but the use of generics in `Problem` made this impossible. This is why I'm thinking about the above so creat [... cut]
PR #37 (2019-06-07): Generation of C/C++ bindings with statically and dynamically linkable libraries 
The extra files and changes needed to generate a c-lib and header.
Currently I have not updated the python scripts though to use the new files.

Closes #8