alphaville/optimization-engine #12: Min/Max constraint for OpEnConstraints in Matlab

BjoernLindqvist, 2019-05-16:
Is your feature request related to a problem? Please describe.
When the descision variables (u1 u2 u3) have different magnitudes the "make_ball" constraints are not really suitable. 

Describe the solution you'd like
An alternative OpEnConstraints that takes Umin and Umax as input vectors.

Describe alternatives you've considered
Tried to implement it myself since the "rectangle.rs" file is already there, but failed miserably due to no knowledge of rust.

korken89, 2019-05-16:
@alphaville Maybe you can have a look at this? I am not yet into the MATLAB part

alphaville, 2019-05-16:
@BjoernLindqvist Τhis is implemented in Rust, but not in MATLAB yet. We will move the code generation to Jinja2 soon, so it will not be worth updating this now. However, I can send you a quick (and dirty) fix tomorrow.

BjoernLindqvist, 2019-05-17:
@alphaville I see. Well I do like dirty fixes. :)

alphaville, 2019-05-21:
@korken89 We need to fix #18 first.

alphaville, 2019-05-27:
@BjoernLindqvist Hi Bjoern. Sorry for the late reply, but we're currently revamping part of the code and moving the code generation functionality to Python.

One (very dirty) way around this issue is to edit the generated code. Look for the line

```rust
let bounds = ...
```

and replace it with:

```rust
let UMIN : Vec<f64> = vec![-1.0; NU];
let UMAX : Vec<f64> = vec![1.0; NU];
let bounds = Rectangle::new(UMIN, UMAX);
```
This will impose the bounds `-1 <= u <= 1`. In order to impose more general bounds, simply write

```rust
let UMIN : Vec<f64> = vec![-1.0, -3.5, 0.0, 5.0, -10.0];
let UMAX : Vec<f64> = vec![1.0, 3.0, 1.0, 2.0, -1.0];
let bounds = Rectangle::new(UMIN, UMAX);
```

Numbers must have a `.0` at the end, otherwise you will get an error.

In order to make MATLAB automatically generate this, you need to do two things: (i) introduce a Rectangle type of constraints, and (ii) edit method `open_impose_bounds` in `matlab/@OpEnOptimizerBuilder`.

**Step i.** Go to `matlab/@OpEnConstraints` and introduce another method like this:

```matlab
function o = make_rectangle(radius, xmin, xmax)
    o = OpEnConstraints();
    o.type = 'rectangle';
    o.params.xmin = xmin;
    o.params.max = xmax;
end
```

**Step ii.** Find  `open_impose_bounds` in `matlab/@OpEnOptimizerBuilder` and change it from 

```matlab
function open_impose_bounds(fid_main, constraints)
switch constraints.get_type()
    case 'ball'
        cstr_params = constraints.get_params();
        if ~(isfield(cstr_params, 'centre') && isempty(cstr_params.centre))
            fprintf(fid_main, '\n\t\tlet bounds = Ball2::new_at_origin_with_radius(%f);\n', cstr_params.radius);
        end
    case 'no_constraints'
        fprintf(fid_main, '\n\t\tlet bounds = NoConstraints::new();\n');
    otherwise
        fprintf(fid_main, '\n\t\tlet bounds = NoConstraints::new();\n');
end
```

to

```matlab
function open_impose_bounds(fid_main, constraints)
switch constraints.get_type()
    case 'ball'
        cstr_params = constraints.get_params();
        if ~(isfield(cstr_params, 'centre') && isempty(cstr_params.centre))
            fprintf(fid_main, '\n\t\tlet bounds = Ball2::new_at_origin_with_radius(%f);\n', cstr_params.radius);
        end
    case 'no_constraints'
        fprintf(fid_main, '\n\t\tlet bounds = NoConstraints::new();\n');
    case 'rectangle'
        % add code here
    otherwise
        fprintf(fid_main, '\n\t\tlet bounds = NoConstraints::new();\n');
end
```

Let me know if you encounter any problems.

alphaville, 2019-06-24:
@BjoernLindqvist @sinashm I just wanted to let you know that we just released a new version of `opengen` - the Python library of Optimization Engine. I think we can close this issue now - I guess you have resolved the issue. If not, please feel free to comment and reopen it or start a new one.

closed 2019-06-24 by alphaville

BjoernLindqvist, 2019-06-26:
@alphaville I'm so sorry, I completely forgot to answer.  Yes, I did my own implementation that works well enough so you can close it. Thank you so much for your help and I will check out the new version! :)

Subhankhan41, 2020-09-21:
@alphaville Hello, I'm trying to add min/max constraints on control input, I'm struggling to understand the inclusion in both the clean way and the dirty way as you have mentioned above. Could you please give me an example where NMPC is considering the constraints on control input.

Thanks