alphaville/optimization-engine #14: Move MATLAB codegen to Jinja2

korken89, 2019-05-16:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Lets move the matlab codegen to Jinja2 templates, the current template code is very hard to read and see how it comes together.
This will also help with the C/C++ lib generation and overview IMO.

**Describe alternatives you've considered**
Keep it as it is.

**Additional context**
None.

alphaville, 2019-09-20:
Jinja is not supported in MATLAB, but there's [Lobster](https://github.com/comkieffer/Lobster-Templates) (but it looks abandoned).

korken89, 2019-09-20:
Matlab can call into python, that should be enough

alphaville, 2019-09-20:
@korken89 that's certainly a legit solution. I'm just concerned that users will need to install the Python package - not too difficult, but the less one does during installation, the better it is.

hampos, 2019-10-30:
Hey guys, you could write your functionality in Java and invoke it from Matlab, since Matlab is natively Java I believe.
This would require your jar in Matlab's classpath though, not sure if you could do this through an add-on.
Cool project btw!