alphaville/optimization-engine #16: Should an optimizer/engine not reset the cache on init?

korken89, 2019-05-18:
As the title states, should not the cache and L-BFGS be reset on init?
Seems like calling solve multiple times can cause issues (I had not time to test this yet).

alphaville, 2019-05-19:
Oh, this seems to be a bug then. Yes, on init the L-BFGS cache should be cleared/reset. 

Another related issue I have in mind is that if an instance of the solver does not converge, the solution might be of poor quality. In that case, we need to decide whether to use it as a warm start for the next instance of the solver or not.

alphaville, 2019-05-21:
Fixed in `clib_impl`; @korken89 by the way, I just merged `feature/penalty-functions` into `clim_impl`. I had some conflicts, but I think I resolved them properly (at least all tests pass).

closed 2019-05-21 by alphaville