alphaville/optimization-engine #17: Build fails on `emil_fixes`

alphaville, 2019-05-20:
**Describe the bug**
Changes on `emil_fixes` make the build on my machine fail.

**To Reproduce**
Check out `emil_fixes` and run `cargo clean` and then `cargo build`. I get the following errors:

```
$ cargo build
   Compiling num-traits v0.2.6                                                                                               
   Compiling num-integer v0.1.39                                                                                             
   Compiling num-bigint v0.2.2                                                                                               
   Compiling num-complex v0.2.1                                                                                              
   Compiling num-iter v0.1.37                                                                                                
   Compiling num-rational v0.2.1                                                                                             
   Compiling num v0.2.0                                                                                                      
   Compiling lbfgs v0.1.1                                                                                                    
   Compiling optimization_engine v0.3.0 (/home/chung/NetBeansProjects/RUST/optimization-engine)                              
error[E0658]: imports can only refer to extern crate names passed with `--extern` on stable channel (see issue #53130)       
 --> src/constraints/mod.rs:7:9                                                                                              
  |                                                                                                                          
3 | mod ball2;                                                                                                               
  | ---------- not an extern crate passed with `--extern`                                                                    
...                                                                                                                          
7 | pub use ball2::Ball2;                                                                                                    
  |         ^^^^^                                                                                                            
  |                                                                                                                          
note: this import refers to the module defined here                                                                          
 --> src/constraints/mod.rs:3:1                                                                                              
  |                                                                                                                          
3 | mod ball2;                                                                                                               
  | ^^^^^^^^^^                                                                                                               
                                                                                                                             
error[E0658]: imports can only refer to extern crate names passed with `--extern` on stable channel (see issue #53130)       
 --> src/constraints/mod.rs:9:9                                                                                              
  |                                                                                                                          
5 | mod rectangle;                                                                                                           
  | -------------- not an extern crate passed with `--extern`                                                                
...                                                                                                                          
9 | pub use rectangle::Rectangle;                                                                                            
  |         ^^^^^^^^^                                                                                                        
  |                                                                                                                          
note: this import refers to the module defined here                                                                          
 --> src/constraints/mod.rs:5:1                                                                                              
  |                                                                                                                          
5 | mod rectangle;                                                                                                           
  | ^^^^^^^^^^^^^^                                                                                                           
                                                                                                                             
error[E0658]: imports can only refer to extern crate names passed with `--extern` on stable channel (see issue #53130)       
  --> src/core/fbs/mod.rs:49:9                                                                                               
   |                                                                                                                         
46 | mod fbs_optimizer;                                                                                                      
   | ------------------ not an extern crate passed with `--extern`                                                           
...                                                                                                                          
49 | pub use fbs_optimizer::FBSOptimizer;                                                                                    
   |         ^^^^^^^^^^^^^                                                                                                   
   |                                                                                                                         
note: this import refers to the module defined here                                                                          
  --> src/core/fbs/mod.rs:46:1                                                                                               
   |                                                                                                                         
46 | mod fbs_optimizer;                                                                                                      
   | ^^^^^^^^^^^^^^^^^^                                                                                                      
                                                                                                                             
error[E0658]: imports can only refer to extern crate names passed with `--extern` on stable channel (see issue #53130)       
 --> src/core/panoc/mod.rs:8:9                                                                                               
  |                                                                                                                          
5 | mod panoc_optimizer;                                                                                                     
  | -------------------- not an extern crate passed with `--extern`                                                          
...                                                                                                                          
8 | pub use panoc_optimizer::PANOCOptimizer;                                                                                 
  |         ^^^^^^^^^^^^^^^                                                                                                  
[... 74 lines cut ...]
   |         ^^^^^^^^^^^^^                                                                                                   
   |                                                                                                                         
note: this import refers to the module defined here                                                                          
[... cut ...]
For more information about this error, try `rustc --explain E0658`.                                                          
error: Could not compile `optimization_engine`.                                                                              

To learn more, run the command again with --verbose.
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**System information:**
 - Ubuntu 14.04
 - Rust: rustc 1.31.0 (abe02cefd 2018-12-04)

**Additional context**
The issue happens only on `emil_fixes`, but not on `master`. Related to pull request #13.

korken89, 2019-05-20:
Hi, you need to update your compiler it seems.

Build fine for me on stable `rustc 1.34.2 (6c2484dc3 2019-05-13)`.

korken89, 2019-05-20:
I have tested and you only need to update one minor version to `rust version 1.32.0 (9fda7c223 2019-01-16)` to have it all working.
So you do not need to go to the latest, only the full Rust 2018.

alphaville, 2019-05-20:
`rustup update` resolved this.

closed 2019-05-20 by alphaville