alphaville/optimization-engine #19: Towards a Python code generator - development plan

alphaville, 2019-05-21:
:construction: The plan is to pause the development of the MATLAB interface and move the code generation functionality to Python. We will then be able to call Python from MATLAB. The idea is to keep all functionality in one place only. Besides, Python is free, while not everyone has access to MATLAB.

At the same time, we will implement the penalty method and later the augmented Lagrangian method too.

Next steps:

- Have a **stable Rust interface** (no major changes in Cache, Engine etc expected); I think we're there
- Write a **minimal working example** of how the generated code will look like in the simple case without the penalty method (we have this from the MATLAB code generator)
- Prepare a minimal working example using the **penalty method**
- Start implementing the existing MATLAB code generator in Python using [Jinja2](http://jinja.pocoo.org/docs/2.10/) (I guess we'll go for Python3, right?)
- Support the penalty method in the code generator

We should be able to **consume** the auto-generated Rust optimizer using its **C bindings** (currently under development).

This will allow us to use it from both Python and MATLAB (by creating a **MEX interface**).

Let's discuss this here below...

korken89, 2019-05-21:
The only comment I have for now is on the Rust interface. 
In `clib_impl` I have made the updates to make constraints take references (and not owning vectors to remove the (de)allocation) which is a breaking change, after this I think we are close to a stable interface.

alphaville, 2019-05-22:
@korken89 Proposal:

[image]

We need to think how to co-design the auto-generated Rust optimizer and the C library. So far we can generated all necessary code using Python (see branch `dev/python`).

korken89, 2019-05-24:
I think this will work out as we were discussing!

alphaville, 2019-06-02:
[Completed](https://alphaville.github.io/optimization-engine/docs/python-interface)

closed 2019-06-02 by alphaville