alphaville/optimization-engine #24: Error using directory handle mkdir

HolyPalladin, 2019-05-24:
Dear Developers,

some days ago, I wanted to experiment with OptimizationEngine OpEn you have developed. My aim was to run the Rust project through the Matlab Interface. After having installed Rust from [https://alphaville.github.io/optimization-engine/docs/installation]. I followed the instructions and downloaded as well the Optimization Engine along with the CasADi Toolbox. I have put the CasADi Toolbox inside the folder "optimization-engine-master" (the folder I get after unzipping the Optimization Engine). After adding the Optimization Engine I ran the following code to activate CasADi:

```
addpath('<yourpath>/casadi-windows-matlabR2016a-v3.4.5')
import casadi.*
x = MX.sym('x')
disp(jacobian(sin(x),x))
```
I wanted to run the example_open_1 and when I stepped in line 9

`opEnOptimizer = opEnBuilder.build();`

I got the following messages.

```
Error using mkdir
The system cannot find the path specified.

Error in OpEnOptimizerBuilder/build>clean_destination (line 153)
    mkdir(destination_dir);

Error in OpEnOptimizerBuilder/build (line 38)
clean_destination(opEnBuilder);

Error in example_open_lv (line 24)
optimizer = builder.build();

```
The destination folder is inside the @OpEnOptimizerBuilder within cd /matlab of the Optimization Engine folder and the current folder is cd /matlab. However, I am not allowed to switch to the destination folder (@OpEnOptimizerBuilder) because it is deemed as a method.

I have the R2019a Matlab version and 64-bit Windows Operating System.

Any help would be appreciated.

alphaville, 2019-05-24:
Thank you for reporting this. The bug comes from line 4 in `matlab_open_root` and it is because the file separator on Windows is the backslash instead of the forward slash. I'll fix it and get back to you.

alphaville, 2019-05-24:
@HolyPalladin  Just before I commit this change, can you check whether the following works: Replace the content of `matlab_open_root.m` with 

```matlab
function open_root = matlab_open_root()

this_file_path = which('matlab_open_root');
matlab_path_str = fileparts(this_file_path);
matlab_path_file = java.io.File(matlab_path_str);
open_root = matlab_path_file.getParent();
```

It works on my Ubuntu system, but I haven't tested it on Windows.

HolyPalladin, 2019-05-24:
@alphaville I performed this change and I get the following message when I run the following line in example_open_1 as also when in the script "matlab_open_root"

`opEnBuilder = OpEnOptimizerBuilder().with_problem(u, p, phi, constraints);
````
Error using fullfile (line 67)
All inputs must be strings, character vectors, or cell arrays of character vectors.

Error in OpEnOptimizerBuilder (line 32)
                .with_build_path(fullfile(optimization_engine_root_path, 'build'))...
```

alphaville, 2019-05-24:
@HolyPalladin Can you share the output of `matlab_open_root()`? Do you get any errors when you run just `matlab_open_root()`?

alphaville, 2019-05-24:
@HolyPalladin I am able to reproduce this. I will get back to you in a while.

alphaville, 2019-05-24:
@HolyPalladin Can you please try again using the following function:

```matlab
function open_root = matlab_open_root()

this_file_path = which('matlab_open_root');
matlab_path_str = fileparts(this_file_path);
matlab_path_file = java.io.File(matlab_path_str);
open_root = char(matlab_path_file.getParent());
```

I just tested it and it works on my system. Hopefully it will work on Windows too (according to MATLAB's documentation, none of the involved commands are OS-specific). Fingers crossed...

Just a side note: you will need to [install the Rust compiler](https://www.rust-lang.org/tools/install) in order to compile the auto-generated code (it's free). Check out OpEn's [installation instructions](https://alphaville.github.io/optimization-engine/docs/installation).

commit d0e1eb16 (2019-05-25): close #24
closed 2019-05-25 by alphaville