alphaville/optimization-engine #26: C file named extern/auto_casadi_cost.c doesn't let me compile icasadi v0.1.2

HolyPalladin, 2019-05-25:
**Describe the bug**
Dear Community,

in my effort in setting the proper format in running OptimizationEngine in Windows through Matlab, along with the hints pointed out yesterday in the previous issue I started, there seems to be some other minor problems. Having changed the content of the matlab_open_root in the Matlab Interface as indicated by @alphaville  to the following:

```
function open_root = matlab_open_root()

this_file_path = which('matlab_open_root');
matlab_path_str = fileparts(this_file_path);
matlab_path_file = java.io.File(matlab_path_str);
open_root = char(matlab_path_file.getParent());
```
solved some issues but then when I compile the example_open_1 document I get the following message:

```
example_open_1
   Compiling cc v1.0.37
   Compiling libc v0.2.55
   Compiling icasadi v0.1.2 (C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi)
error: failed to run custom build command for `icasadi v0.1.2 (C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi)`
process didn't exit successfully: `C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi\target\release\build\icasadi-4c7d82453aee18ad\build-script-build` (exit code: 1)
--- stdout
TARGET = Some("x86_64-pc-windows-msvc")
OPT_LEVEL = Some("3")
HOST = Some("x86_64-pc-windows-msvc")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
DEBUG = Some("false")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
running: "C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX64\\x64\\cl.exe" "/nologo" "/MD" "/O2" "/I" "src" "/W4" "-Wno-unused-parameter" "-Wall" "/FoC:\\optimization-engine-master\\optimization-engine-master\\build\\autogenerated_optimizer\\icasadi\\target\\release\\build\\icasadi-7eaa844b72b47fd5\\out\\extern\\auto_casadi_cost.o" "/c" "extern/auto_casadi_cost.c"
cargo:warning=cl : Command line error D8021 : invalid numeric argument '/Wno-unused-parameter' 
exit code: 2

--- stderr

error occurred: Command "C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX64\\x64\\cl.exe" "/nologo" "/MD" "/O2" "/I" "src" "/W4" "-Wno-unused-parameter" "-Wall" "/FoC:\\optimization-engine-master\\optimization-engine-master\\build\\autogenerated_optimizer\\icasadi\\target\\release\\build\\icasadi-7eaa844b72b47fd5\\out\\extern\\auto_casadi_cost.o" "/c" "extern/auto_casadi_cost.c" with args "cl.exe" did not execute successfully (status code exit code: 2).

[codegen] Build command: cargo build
   Compiling autocfg v0.1.4
   Compiling num-bigint v0.2.2
   Compiling proc-macro2 v0.4.30
   Compiling unicode-xid v0.1.0
   Compiling num-rational v0.2.1
   Compiling num-complex v0.2.1
   Compiling syn v0.15.34
   Compiling libc v0.2.55
   Compiling serde v1.0.91
   Compiling cc v1.0.37
   Compiling ryu v0.2.8
   Compiling itoa v0.4.4
   Compiling num-traits v0.2.8
   Compiling num-integer v0.1.41
   Compiling num-iter v0.1.39
   Compiling icasadi v0.1.2 (C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi)
   Compiling quote v0.6.12
error: failed to run custom build command for `icasadi v0.1.2 (C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi)`
process didn't exit successfully: `C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\target\debug\build\icasadi-ff7105e17f623f1b\build-script-build` (exit code: 1)
--- stdout
TARGET = Some("x86_64-pc-windows-msvc")
OPT_LEVEL = Some("0")
HOST = Some("x86_64-pc-windows-msvc")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
DEBUG = Some("true")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-pc-windows-msvc = None
CC_x86_64_pc_windows_msvc = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-msvc = None
CFLAGS_x86_64_pc_windows_msvc = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
running: "C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX64\\x64\\cl.exe" "/nologo" "/MD" "/Z7" "/I" "src" "/W4" "-Wno-unused-parameter" "-Wall" "/FoC:\\optimization-engine-master\\optimization-engine-master\\build\\autogenerated_optimizer\\target\\debug\\build\\icasadi-081876f384fb7888\\out\\extern\\auto_casadi_cost.o" "/c" "extern/auto_casadi_cost.c"
cargo:warning=cl : Command line error D8021 : invalid numeric argument '/Wno-unused-parameter' 
exit code: 2

--- stderr

error occurred: Command "C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX64\\x64\\cl.exe" "/nologo" "/MD" "/Z7" "/I" "src" "/W4" "-Wno-unused-parameter" "-Wall" "/FoC:\\optimization-engine-master\\optimization-engine-master\\build\\autogenerated_optimizer\\target\\debug\\build\\icasadi-081876f384fb7888\\out\\extern\\auto_casadi_cost.o" "/c" "extern/auto_casadi_cost.c" with args "cl.exe" did not execute successfully (status code exit code: 2).

warning: build failed, waiting for other jobs to finish...
error: build failed
Warning: Unsuccessful read:  The specified amount of data was not returned within the Timeout period.
'udp' unable to read any data. For more information on possible reasons, see UDP Read Warnings. 
Error using jsondecode
JSON syntax error: expected value but found end of text.

Error in OpEnOptimizer/consume (line 59)
            out = jsondecode(json_response);

Error in example_open_1 (line 14)
out = opEnOptimizer.consume([1,2]);

```

To me the issue is with the part of the error mentioning that:

`extern\\auto_casadi_cost.o" "/c" "extern/auto_casadi_cost.c" with args "cl.exe" did not execute successfully (status code exit code: 2).`

I am using Windows 64-bit OS and the C compiler is MSVC in Matlab R2019a version and the most recent version of Rust and CasADi.

Any help would be appreciated.

Thank you.

alphaville, 2019-05-25:
It seems to me that the compilation fails because of this issue: `cargo:warning=cl : Command line error D8021 : invalid numeric argument '/Wno-unused-parameter'`. I think that this is a GCC-specific flag. 

Can you go to `build/autogenerated_optimizer/icasadi/src/build.rs` and remove line 8:

```rust
.flag("-Wno-unused-parameter")
```

and try again to see what happens.

alphaville, 2019-05-25:
We found the culprit - @korken89 can we check whether the user is compiling with gcc and have compiler-specific flags? Or remove `-Wno-unused-parameter`?

commit f0d9ae91 (2019-05-25): close #26; close #27
closed 2019-05-25 by alphaville