alphaville/optimization-engine #27: Cargo build fails with MSVC on Windows version II

HolyPalladin, 2019-05-25:
I removed from \optimization-engine-master \icasadi\src\build.rs line 8 as well as the same line from the \optimization-engine-master\build\autogenerated_optimizer\icasadi\src\build.rs and ran the example_open_1 again and this time I ended up with the following message:

```
Compiling icasadi v0.1.2 (C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi)
    Finished release [optimized] target(s) in 2.60s
[codegen] Build command: cargo build
   Compiling icasadi v0.1.2 (C:\optimization-engine-master\optimization-engine-master\build\autogenerated_optimizer\icasadi)
   Compiling optimization_engine v0.3.1 (C:\optimization-engine-master\optimization-engine-master)
error[E0463]: can't find crate for `num`
  --> C:\optimization-engine-master\optimization-engine-master\src\lib.rs:15:1
   |
15 | extern crate num;
   | ^^^^^^^^^^^^^^^^^ can't find crate

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: Could not compile `optimization_engine`.
warning: build failed, waiting for other jobs to finish...
error: build failed
Warning: Unsuccessful read:  The specified amount of data was not returned within the Timeout period.
'udp' unable to read any data. For more information on possible reasons, see UDP Read Warnings. 
Error using jsondecode
JSON syntax error: expected value but found end of text.

Error in OpEnOptimizer/consume (line 59)
            out = jsondecode(json_response);

Error in example_open_1 (line 14)
out = opEnOptimizer.consume([1,2]);
```

Also when I ran the command rustup show in the command window I got:

```
stable-x86_64-pc-windows-msvc (default)
rustc 1.34.1 (fc50f328b 2019-04-24)
```

alphaville, 2019-05-25:
I wonder whether this is related to [this issue](https://github.com/rust-lang/cargo/issues/4072). Can you try to edit `Cargo.toml` in `optimization-engine/` and add the following lines:

```
[http]
check-revoke = true
```

alphaville, 2019-05-25:
Update: @HolyPalladin sent me the following output of `cargo build -v`:

```
C:\optimization-engine-master\optimization-engine-master>cargo build -v
Fresh autocfg v0.1.4
Fresh num-traits v0.2.8
Fresh num-integer v0.1.41
Fresh num-complex v0.2.1
Fresh num-bigint v0.2.2
Fresh num-iter v0.1.39
Fresh num-rational v0.2.1
Fresh num v0.2.0
Fresh lbfgs v0.1.1
Fresh optimization_engine v0.3.1
Compiling optimization_engine v0.3.1 (C:\optimization-engine-master\optimization-engine-master)
Running `rustc --edition=2018 
--crate-name optimization_engine src\lib.rs 
--color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C 
metadata=8bc95cdf3be1a96d -C extra-filename=-8bc95cdf3be1a96d 
--out-dir C:\optimization-engine-master\optimization-engine-master\target\debug\deps 
-C 
incremental=C:\optimization-engine-master\optimization-engine-master\target\debug\incremental 
-L dependency=C:\optimization-engine-master\optimization-engine-master\target\debug\deps 
--extern lbfgs=C:\optimization-engine-master\optimization-engine-master\target\debug\deps\liblbfgs-4bc74265522bdee5.rlib 
--extern optimization_engine=C:\optimization-engine-master\optimization-engine-master\target\debug\deps\liboptimization_engine-2df575398fc29577.rlib`

error[E0463]: can't find crate for `num`
--> src\lib.rs:15:1
|
15 | extern crate num;
| ^^^^^^^^^^^^^^^^^ can't find crate

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: Could not compile `optimization_engine`.

Caused by:
process didn't exit successfully: `rustc --edition=2018 
--crate-name optimization_engine src\lib.rs 
--color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C 
metadata=8bc95cdf3be1a96d -C extra-filename=-8bc95cdf3be1a96d 
--out-dir C:\optimization-engine-master\optimization-engine-master\target\debug\deps 
-C 
incremental=C:\optimization-engine-master\optimization-engine-master\target\debug\incremental 
-L dependency=C:\optimization-engine-master\optimization-engine-master\target\debug\deps 
--extern lbfgs=C:\optimization-engine-master\optimization-engine-master\target\debug\deps\liblbfgs-4bc74265522bdee5.rlib 
--extern optimization_engine=C:\optimization-engine-master\optimization-engine-master\target\debug\deps\liboptimization_engine-2df575398fc29577.rlib` (exit code: 1)
```

alphaville, 2019-05-25:
Problem solved: it turns out that this is because of the following link in the auto-generated `Cargo.toml`:

```
optimization_engine = {path = "../../"}
```

changing this to

```
optimization_engine = "0.3.1"
```

fixes the issue (no idea why).

commit f0d9ae91 (2019-05-25): close #26; close #27
closed 2019-05-25 by alphaville