alphaville/optimization-engine #46: Multiple solvers in one final binary

korken89, 2019-06-07:
**Is your feature request related to a problem? Please describe.**
Not really, but it is currently very difficult to use multiple `icasadi` instances of generated solvers as the linking from `casadi` C code is not using unique names (it currently uses the name given by the MATLAB/Python `casadi` interface.
This causes linker errors as symbols are multiply defined. 

When manually creating `icasadi` functions this is not a problem, as the user can take any name.

**Describe the solution you'd like**
The auto-generated C code needs to get unique names.

**Describe alternatives you've considered**
1. Leave as is, and then it is difficult to define multiple solvers in the same binary.
2. Extent code generation to add a random identifier to user names of `casadi` functions and auto-generate the `icasadi` file that will interface to the generated C code.

**Additional context**
None.

alphaville, 2019-06-07:
Wouldn't it make sense to generate two different (auto-generated) solvers in that case. These would have different names and would use different (local) `icasadi` crates. Would that be a problem? 

Alternatively, we need an interface that can support code generation for multiple problems and solver configurations.

For what it's worth, the C file names in icasadi are configurable: in Python, see `BuildConfiguration.__cost_function_name` and similar fields.

korken89, 2019-06-08:
I think this should be fine, as long as all C-linkable names are unique (i.e. adding a random string to their names is the usual fix). 
On top of this we need to make a small change in the C/C++ interface generation so those are named after the solver you create where now all are called `solver_new `, etc where it should be `{name_from_python_generator}_new`, etc.

No big changes, but then we have secured this part of the generation :)

PR #51 (2019-06-22): Now with unique names in C based on solver name
Fixes #46
closed 2019-06-22 by korken89