alphaville/optimization-engine #54: MATLAB OpEN ncp problem

sinashm, 2019-06-14:
The Matlab interface had following error:

Undefined function or variable 'ncp'.

Error in casadi_generate_c_code (line 68)
fprintf(fid, '#define CASADI_NUM_CONSTAINTS_TYPE_PENALTY %d\n', ncp);

Error in OpEnOptimizerBuilder/build>casadi_codegen (line 98)
[cost, grad_cost] = casadi_generate_c_code(u, p, phi,cstr_pen,build_directory);

Error in OpEnOptimizerBuilder/build (line 54)
casadi_codegen(opEnBuilder.sx_u, opEnBuilder.sx_p, opEnBuilder.cost, ...

Error in optimization_build (line 57)
optimizer = builder.build();

Error in run (line 47)
optimization_build

alphaville, 2019-06-14:
This is now fixed in PR #55. Will be merged soon into `master`.

korken89, 2019-06-14:
Merged

closed 2019-06-14 by korken89