alphaville/optimization-engine #58: Python Example Error

sinashm, 2019-06-21:
I am trying your example from webpage and I am getting following error:

Traceback (most recent call last):
File "Example.py", line 35, in
builder.build()
File "/usr/local/lib/python2.7/dist-packages/opengen/builder/optimizer_builder.py", line 347, in build
self.__build_optimizer() # build overall project
File "/usr/local/lib/python2.7/dist-packages/opengen/builder/optimizer_builder.py", line 260, in __build_optimizer
raise Exception('Rust build failed')
Exception: Rust build failed

+Rust is installed

    The code:
    import casadi.casadi as cs
    import opengen as og

u = cs.SX.sym("u", 5) # decision variable (nu = 5)
p = cs.SX.sym("p", 2) # parameter (np = 2)
phi = og.functions.rosenbrock(u, p) # cost function

bounds = og.constraints.Ball2(None, 1.5) # ball centered at origin

problem = og.builder.Problem(u, p, phi).with_constraints(bounds)

meta = og.config.OptimizerMeta()
.with_version("0.0.0")
.with_authors(["P. Sopasakis", "E. Fresk"])
.with_licence("CC4.0-By")
.with_optimizer_name("the_optimizer")

build_config = og.config.BuildConfiguration()
.with_build_directory("python_build")
.with_build_mode("debug")

solver_config = og.config.SolverConfiguration()
.with_lfbgs_memory(15)
.with_tolerance(1e-5)
.with_max_inner_iterations(155)

builder = og.builder.OpEnOptimizerBuilder(problem,
metadata=meta,
build_configuration=build_config,
solver_configuration=solver_config)
builder.enable_tcp_interface()
builder.build()

mng = og.tcp.OptimizerTcpManager('python_build/the_optimizer')
mng.start()

pong = mng.ping() # check if the server is alive
print(pong)
solution = mng.call([1.0, 50.0]) # call the solver over TCP
print(solution)

mng.kill()

**Describe the bug**

A clear and concise description of what the bug is.

**To Reproduce**

Steps to reproduce the behavior:
1. Step 1
2. Step 2.
3. See error

**Expected behavior**

A clear and concise description of what you expected to happen.

**System information:**

:warning: Please, provide the following information:

 - System/Platform [e.g., Intel i5, Raspeberry Pi Model B, Mobile device, etc]
 - OS: [e.g. Ubuntu, iOS, Windows, Raspbian, etc]
 - What is the output of `rustup show`?
 - What is the output of `rustc -V`?
 - Python/MATLAB version if relevant

**Additional context**

Add any other context about the problem here.

korken89, 2019-06-22:
You seem to have used the released version of OpEn to build the development version. Please wait a day or so and we will release the new version, or do the following to install the development version:

1. `sudo pip3 uninstall opengen`
2. `cd open-codegen/opengen`
3. `sudo pip3 install .`

Here is a reference python file (you also seem to have missed the `\` in the code), tested with commit 64977f40a83427405711778f1ae7861b1e473b38:

```python
import casadi.casadi as cs
import opengen as og

u = cs.SX.sym("u", 5) # decision variable (nu = 5)
p = cs.SX.sym("p", 2) # parameter (np = 2)
phi = og.functions.rosenbrock(u, p) # cost function

bounds = og.constraints.Ball2(None, 1.5) # ball centered at origin

problem = og.builder.Problem(u, p, phi).with_constraints(bounds)

meta = og.config.OptimizerMeta() \
    .with_version("0.0.0") \
    .with_authors(["P. Sopasakis", "E. Fresk"]) \
    .with_licence("CC4.0-By") \
    .with_optimizer_name("the_optimizer")

build_config = og.config.BuildConfiguration() \
    .with_build_directory("python_build") \
    .with_build_mode("debug")

solver_config = og.config.SolverConfiguration() \
    .with_lfbgs_memory(15) \
    .with_tolerance(1e-5) \
    .with_max_inner_iterations(155)

builder = og.builder.OpEnOptimizerBuilder(problem,
    metadata=meta,
    build_configuration=build_config,
    solver_configuration=solver_config)
builder.enable_tcp_interface()
builder.build()

mng = og.tcp.OptimizerTcpManager('python_build/the_optimizer')
mng.start()

pong = mng.ping() # check if the server is alive
print(pong)
solution = mng.call([1.0, 50.0]) # call the solver over TCP
print(solution)

mng.kill()
```

alphaville, 2019-06-22:
Hi @sinashm. Can you please provide your system information as in the **bug template**? Which version of `opengen` are you using? Did you download it using `pip install opengen`? What OS are you using? 

I think that the error you're getting says that the code has been generated, but rust cannot compile it. In order to get a better understanding of the error, can you go inside the auto-generated folder (that is, `python_build/the_optimizer`) and run the following command:

```console
cargo build
```

As a side note, indentations are important in Python. Can you double-check that you have copied the example code properly?

sinashm, 2019-06-22:
Hi @alphaville,

I am running Ubuntu 16.04, Python 2.7.12 and I used sudo pip install opengen for installation. The version of opengen is opengen-0.0.7
The output of cargo build is:

sina:the_optimizer$ cargo build
   Compiling serde_json v1.0.39
   Compiling icasadi v0.2.1 (/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/icasadi)
error: failed to run custom build command for `icasadi v0.2.1 (/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/icasadi)`
process didn't exit successfully: `/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-ed0abb2625a5001c/build-script-build` (exit code: 101)
--- stdout
TARGET = Some("x86_64-unknown-linux-gnu")
OPT_LEVEL = Some("0")
HOST = Some("x86_64-unknown-linux-gnu")
CC_x86_64-unknown-linux-gnu = None
CC_x86_64_unknown_linux_gnu = None
HOST_CC = None
CC = None
CFLAGS_x86_64-unknown-linux-gnu = None
CFLAGS_x86_64_unknown_linux_gnu = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
DEBUG = Some("true")
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-unknown-linux-gnu = None
CC_x86_64_unknown_linux_gnu = None
HOST_CC = None
CC = None
CFLAGS_x86_64-unknown-linux-gnu = None
CFLAGS_x86_64_unknown_linux_gnu = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-unknown-linux-gnu = None
CC_x86_64_unknown_linux_gnu = None
HOST_CC = None
CC = None
CFLAGS_x86_64-unknown-linux-gnu = None
CFLAGS_x86_64_unknown_linux_gnu = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-unknown-linux-gnu = None
CC_x86_64_unknown_linux_gnu = None
HOST_CC = None
CC = None
CFLAGS_x86_64-unknown-linux-gnu = None
CFLAGS_x86_64_unknown_linux_gnu = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
CC_x86_64-unknown-linux-gnu = None
CC_x86_64_unknown_linux_gnu = None
HOST_CC = None
CC = None
CFLAGS_x86_64-unknown-linux-gnu = None
CFLAGS_x86_64_unknown_linux_gnu = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2")
running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-I" "src" "-Wall" "-Wextra" "-Wall" "-Wpedantic" "-Wno-long-long" "-Wno-unused-parameter" "-o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/auto_casadi_cost.o" "-c" "extern/auto_casadi_cost.c"
exit code: 0
running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-I" "src" "-Wall" "-Wextra" "-Wall" "-Wpedantic" "-Wno-long-long" "-Wno-unused-parameter" "-o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/auto_casadi_grad.o" "-c" "extern/auto_casadi_grad.c"
exit code: 0
running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-I" "src" "-Wall" "-Wextra" "-Wall" "-Wpedantic" "-Wno-long-long" "-Wno-unused-parameter" "-o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/auto_casadi_constraints_type_penalty.o" "-c" "extern/auto_casadi_constraints_type_penalty.c"
exit code: 0
running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-I" "src" "-Wall" "-Wextra" "-Wall" "-Wpedantic" "-Wno-long-long" "-Wno-unused-parameter" "-o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/icasadi.o" "-c" "extern/icasadi.c"
exit code: 0
AR_x86_64-unknown-linux-gnu = None
AR_x86_64_unknown_linux_gnu = None
HOST_AR = None
AR = None
running: "ar" "crs" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/libicasadi.a" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/auto_casadi_cost.o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/auto_casadi_grad.o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/auto_casadi_constraints_type_penalty.o" "/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out/extern/icasadi.o"
exit code: 0
cargo:rustc-link-lib=static=icasadi
cargo:rustc-link-search=native=/home/sina/work_spaces/codes/Python_OpEn/python_build/the_optimizer/target/debug/build/icasadi-be6cc7b01cf3f3b0/out

--- stderr
thread 'main' panicked at 'Unable to find libclang: "couldn\'t find any valid shared libraries matching: [\'libclang.so\', \'libclang-*.so\', \'libclang.so.*\'], set the `LIBCLANG_PATH` environment variable to a path where one of these files can be found (invalid: [])"', src/libcore/result.rs:997:5
note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.

warning: build failed, waiting for other jobs to finish...
error: build failed

I tried the code that Emil shared above and got same error.

sinashm, 2019-06-22:
> You seem to have used the released version of OpEn to build the development version. Please wait a day or so and we will release the new version, or do the following to install the development version:
> 
>     1. `sudo pip3 uninstall opengen`
> 
>     2. `cd open-codegen`
> 
>     3. `sudo pip3 install .`
> 
> 
> Here is a reference python file (you also seem to have missed the `\` in the code), tested with commit [64977f4](https://github.com/alphaville/optimization-engine/commit/64977f40a83427405711778f1ae7861b1e473b38):
> 
> ```python
> import casadi.casadi as cs
> import opengen as og
> 
> u = cs.SX.sym("u", 5) # decision variable (nu = 5)
> p = cs.SX.sym("p", 2) # parameter (np = 2)
> phi = og.functions.rosenbrock(u, p) # cost function
> 
> bounds = og.constraints.Ball2(None, 1.5) # ball centered at origin
> 
> problem = og.builder.Problem(u, p, phi).with_constraints(bounds)
> 
> meta = og.config.OptimizerMeta() \
>     .with_version("0.0.0") \
>     .with_authors(["P. Sopasakis", "E. Fresk"]) \
>     .with_licence("CC4.0-By") \
>     .with_optimizer_name("the_optimizer")
> 
> build_config = og.config.BuildConfiguration() \
>     .with_build_directory("python_build") \
>     .with_build_mode("debug")
> 
> solver_config = og.config.SolverConfiguration() \
>     .with_lfbgs_memory(15) \
>     .with_tolerance(1e-5) \
>     .with_max_inner_iterations(155)
> 
> builder = og.builder.OpEnOptimizerBuilder(problem,
>     metadata=meta,
>     build_configuration=build_config,
>     solver_configuration=solver_config)
> builder.enable_tcp_interface()
> builder.build()
> 
> mng = og.tcp.OptimizerTcpManager('python_build/the_optimizer')
> mng.start()
> 
> pong = mng.ping() # check if the server is alive
> print(pong)
> solution = mng.call([1.0, 50.0]) # call the solver over TCP
> print(solution)
> 
> mng.kill()
> ```

@korken89 I followed your steps and it seems working,
I am getting a solution.

DEBUG:root:Sending request to TCP/IP server
{u'exit_status': u'Converged', u'max_constraint_violation': 0.0, u'num_outer_iterations': 1, u'solution': [0.9064781646490043, 0.8254860578824491, 0.6864387632940303, 0.475070417629115, 0.22354094609769304], u'last_problem_norm_fpr': 6.928368508833967e-06, u'solve_time_ms': 2.332177, u'num_inner_iterations': 28}
INFO:root:Killing server

alphaville, 2019-06-22:
@korken89 @sinashm This is still strange because we had tested the released version of opengen on Ubuntu and it links to the previously released version of OpEn in Rust (0.4.0), so it should have worked. In any case, I think we can close this issue.

closed 2019-06-22 by alphaville