alphaville/optimization-engine #64: Changing the cost function in Python does not update the generated solver after build

korken89, 2019-06-27:
Changing the cost function in Python does not update the generated solver after build, some build artifacts are not detected as old and rebuilt.
We need to look into this issue more, not 100% sure where the issue is yet.

Mitigation: Enable the `.with_rebuild(True)`

alphaville, 2019-06-30:
I cannot confirm the issue on my system (Ubuntu). On what operating system does this happen? Perhaps we need to write some tests in Python and run them on Travis. Can you send me the exact script you're running and provide some more details (e.g., what it prints the first and second time you run it)?

korken89, 2019-07-02:
I have to look more into this.

alphaville, 2019-07-20:
I did the following and I wasn't able to confirm the issue. First, I ran the following script:

```python
import opengen as og
import casadi.casadi as cs

# Build parametric optimizer
# ------------------------------------
u = cs.SX.sym("u", 5)
p = cs.SX.sym("p", 2)
phi = og.functions.rosenbrock(u, p)
c = cs.vertcat(1.9 * u[0] - u[1],
               cs.fmax(0.0, u[2] - u[3] + 0.1))
bounds = og.constraints.Ball2(None, 1.5)
problem = og.builder.Problem(u, p, phi) \
    .with_penalty_constraints(c)        \
    .with_constraints(bounds)
build_config = og.config.BuildConfiguration()  \
    .with_build_directory("python_test_build") \
    .with_build_mode("debug")                  \
    .with_tcp_interface_config()
meta = og.config.OptimizerMeta()                   \
    .with_optimizer_name("tcp_enabled_optimizer")
solver_config = og.config.SolverConfiguration()    \
    .with_tolerance(1e-5)                          \
    .with_constraints_tolerance(1e-4)
builder = og.builder.OpEnOptimizerBuilder(problem, meta,
                                          build_config, solver_config)
builder.build()

mng = og.tcp.OptimizerTcpManager('python_test_build/tcp_enabled_optimizer')
mng.start()

mng.ping()
out = mng.call([1.0, 50.0])
print(out['solution'])
mng.kill()
```

The solution was

```js
[0.015494056386313704, 
 0.029416452881444014, 
-0.03129591706343518, 
 0.068619584751003, 
 0.004235490780720795]
```

I then changed the constraints as follows:

```python
c = cs.vertcat(1.7 * u[0] - u[1],
               cs.fmax(0.0, u[2] - u[3] + 0.1))
```

The solution became:

```js
[0.01818908949406206, 
 0.03090179639810972, 
-0.031239994352222484, 
 0.06868210044117966, 
 0.004214399813686539]
```

Lastly, I changed the cost function to `phi = cs.sumsqr(u)` and the solution became:

```js
[6.460145241379322e-13, 
 6.1317436684975065e-12,
-0.049959935053947385, 
 0.04996010457912201, 
 4.099992133549515e-12]
```
Everything seems to work like clockwork.

korken89, 2019-07-23:
Okey, lets keep it alive if it surfaces again, I am also having issues reproducing it.