alphaville/optimization-engine #74: Bug in Solver

sinashm, 2019-08-17:
I am getting the following error:

thread '<unnamed>' panicked at 'index out of bounds: the len is 57 but the index is 7453010376530526208', /home/sina/.cargo/registry/src/github.com-1ecc6299db9ec823/optimization_engine-0.5.0/src/continuation/homotopy_optimizer.rs:164:13
note: Run with RUST_BACKTRACE=1 environment variable to display a backtrace.
fatal runtime error: failed to initiate panic, error 5
Aborted (core dumped)

on Ubuntu 16.04
with rustc 1.35.0

I am using binder of generated code in ROS, for regenerating the code, you need rosbag of the data.

alphaville, 2019-08-17:
@sinashm can you please provide some context? It would be useful to know what code you were trying to run.

sinashm, 2019-08-17:
@alphaville it is ROS node, i added to the github of the rosnode, 
I will also share with you the python code that I used to generate the C code

alphaville, 2019-08-17:
@sinashm Yes, I would like to have a look at the Python code that you use to generate the optimizer. I got your invitation to project `LTU-CEG/Python_OpEn_MAV`, but folder `floating_object` seems to be a submodule to which I don't have access. Could you paste your Python code here for completeness?

From what I can tell, the error comes from `initialize_param` in `homotopy_optimizer` where the penalty parameters are initialized.

sinashm, 2019-08-17:
I appreciate if you check the code in the following:

import casadi.casadi as cs
import opengen as og
import numpy as np
import math

N=10
dt=1.0/10.0
nu=3 #number of inputs
nx=6 #number of states
no=4 #number of obstacles
nd=1 #number of safety distance
cost=0;
u = cs.SX.sym("u", N*nu) # decision variable (nu = 3 T theta_x theta_y)
p = cs.SX.sym("p", nx+nx+no+nd) # states + state_ref (np = 8 p p_dot theta_x theta_y) + obstacles
X= p[0:nx]
x_ref = p[nx:nx+nx]
x_obst= p[nx+nx:nx+nx+no] # 0: +x 1:-x 2:+y 3: -y 
d_safe= p[nx+nx+no:] # [m]

## weights and hovering reference
u_hover=[9.81,0,0]
Q_state=[10.0,5.0,5.0,5.0,1.0,1.0];
Q_hovering=[1.0,1.0,1.0];
Q_u=[1,1,1];
Q_obstacle=[5.0,5.0,5.0,5.0];  # 0: +x 1:-x 2:+y 3: -y

## Constants
g=9.81 # %gravity
mass = 1.1# % of platform, in kilograms
max_thrust = 30# % of platform, in Newtons

a_x=0.1# % Mass normalized drag coefficients
a_y=0.1#
a_z=0.2#

tao_roll=0.5 #; % Time constants in seconds
tao_pitch=0.5 #;

k_roll=1 ##; % Angle gains
k_pitch=1 #;
x0=X

# bounds on input
u_min = [0,-math.pi/10,-math.pi/10]
u_max = [30,math.pi/10,math.pi/10]
umin=np.kron(np.ones((1,N)),u_min)
umin=umin.tolist()
umin=umin[0]
umax=np.kron(np.ones((1,N)),u_max)
umax=umax.tolist()
umax=umax[0]
c=[]

for i in range(N) :
    k = (i ) * nu
    u0=u[k:k+nu]
    p_0=x0[0]#; % position vector
    v=x0[1:4]#; % velocity vector
    roll=x0[4]#; % roll angle
    pitch=x0[5]#; % pitch angle
    T = u0[0]#; % mass normalized thrust
    roll_g = u0[1]#; %desired angles
    pitch_g = u0[2]#;
    #R_x = np.matrix([[1, 0, 0],[ 0, np.cos(roll), -np.sin(roll)],[0, np.sin(roll), np.cos(roll)]])#; %Rotation around x (roll)
    #R_y = np.matrix([[np.cos(pitch), 0, np.sin(pitch)],[ 0, 1, 0],[-np.sin(pitch), 0, np.cos(pitch)]])#; %Rotation around y (pitch)
    #R = R_y*R_x#; % Rotation matrix without rotation around z since coordinates are fixed to the frame
    #x_0 = p_0[0]
    #y_0 = p_0[1]
    #z_0 = p_0[2]
    #v_dot = R*np.matrix([[0],[ 0],[T]]) + np.matrix([[0],[ 0],[-g]]) - np.matrix([[a_x, 0, 0],[ 0,a_y, 0],[ 0, 0, a_z]])*v
    ddx= (cs.sin(pitch)*cs.cos(roll))*T-a_x*x0[1]
    ddy= (-cs.sin(roll))*T-a_y*x0[2]
    ddz=-g+(cs.cos(pitch)*cs.cos(roll))*T-a_z*x0[3]

    roll_dot = (1/tao_roll)*(k_roll*roll_g - roll)

    pitch_dot = (1/tao_pitch)*(k_pitch*pitch_g - pitch)

    #x0[0]+=x0[3]*dt
    #x0[1]+=x0[4]*dt
    x0[0]+=x0[3]*dt
    x0[1]+=ddx*dt
    x0[2]+=ddy*dt
    x0[3]+=ddz*dt
    x0[4]+=roll_dot*dt
    x0[5]+=pitch_dot*dt
    # Obstacle dynamics
    x_obst[0]+=(-x0[1])*dt
    x_obst[1]+=(x0[1])*dt
    x_obst[2]+=(-x0[2])*dt
    x_obst[3]+=(x0[2])*dt
    #x0=[[x],[y],[z],[v_x],[v_y],[v_z],[roll],[pitch]]
    #cost= np.power((x0[0]-x_ref[0]),2)+np.power((x0[1]-x_ref[1]),2)+np.power((x0[2]-x_ref[2]),2)+np.power((x0[3]-x_ref[3]),2)+np.power((x0[4]-x_ref[4]),2)+np.power((x0[5]-x_ref[5]),2) + cost
    #print(Q_state)
    cost+= cs.sumsqr(Q_state*(x0-x_ref))+cs.sumsqr(Q_hovering*(u0-u_hover))
    #A=x0-x_ref
    #B=Q_state*A
    #print(A[5])
    #print(B[5])
    #obstacle avoidance from 2D lidar
    #cost+= Q_obstacle[0]*cs.fmax(d_safe-x_obst[0]+x0[3]*dt,0) #+x
    #cost+= Q_obstacle[1]*cs.fmax(0.0,d_safe-x_obst[1]-x0[3]*dt) #-x
    #cost+= Q_obstacle[2]*cs.fmax(d_safe-x_obst[2]+x0[4]*dt,0) #+y
    #cost+= Q_obstacle[3]*cs.fmax(d_safe-x_obst[3]-x0[4]*dt,0) #-y
    #print(x0[5])
    if i < N - 1:
        cost+=cs.sumsqr(Q_u*(u0-u[(i+1)*nu:(i+1)*nu+nu]))
    # Constraints
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[0]*(d_safe-x_obst[0]-(-x0[1])*dt))) #+x vx
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[1]*(d_safe-(x_obst[1])-x0[1]*dt))) #-x -vx
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[2]*(d_safe-x_obst[2]-x0[2]*dt))) #+y vy
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[3]*(d_safe-(x_obst[3])-(-x0[2])*dt))) #-y -vy

#c = cs.vertcat(cs.fmax(0.0, -u[0] ),
#               cs.fmax(math.pi/2,  u[1] ),
#               cs.fmax(math.pi/2, -u[1] ),
#               cs.fmax(math.pi/2,  u[2] ),
#               cs.fmax(math.pi/2, -u[2] ),
#                 )

bounds= og.constraints.Rectangle(umin,umax)
problem = og.builder.Problem(u, p, cost) \
.with_penalty_constraints(c)  \
.with_constraints(bounds)
          
          
meta = og.config.OptimizerMeta() \
    .with_version("0.0.0") \
    .with_authors(["P. Sopasakis", "E. Fresk"]) \
    .with_licence("CC4.0-By") \
    .with_optimizer_name("the_optimizer") 

build_config = og.config.BuildConfiguration() \
    .with_build_directory("python_build") \
    .with_build_mode("release") \
     .with_build_c_bindings()  \
    .with_rebuild(True)

solver_config = og.config.SolverConfiguration() \
    .with_lfbgs_memory(15) \
    .with_tolerance(1e-5) \
    .with_max_inner_iterations(155)

builder = og.builder.OpEnOptimizerBuilder(problem,
    metadata=meta,
    build_configuration=build_config,
    solver_configuration=solver_config)
builder.enable_tcp_interface()
builder.build()

mng = og.tcp.OptimizerTcpManager('python_build/the_optimizer')
mng.start()

pong = mng.ping() # check if the server is alive
print(pong)
x_init= [0.0,0.0,0.0,0.0,0.0,0.0];
x_ref = [10.0,0.0,0.0,0.0,0.0,0.0];
x_obst = [0.0,3.0,0.0,0.0];
d_safe=[2.0];
x_0=x_init+x_ref + x_obst +d_safe  #concatenate lists in Python
solution = mng.call(x_0) # call the solver over TCP
print(solution)
print('u_0:')
u0=solution[u'solution']
print(u0[0:3])

mng.kill()

sinashm, 2019-08-17:
Please kindly find the code in the following:

`import casadi.casadi as cs
import opengen as og
import numpy as np
import math

N=10
dt=1.0/10.0
nu=3 #number of inputs
nx=6 #number of states
no=4 #number of obstacles
nd=1 #number of safety distance
cost=0;
u = cs.SX.sym("u", N*nu) # decision variable (nu = 3 T theta_x theta_y)
p = cs.SX.sym("p", nx+nx+no+nd) # states + state_ref (np = 8 p p_dot theta_x theta_y) + obstacles
X= p[0:nx]
x_ref = p[nx:nx+nx]
x_obst= p[nx+nx:nx+nx+no] # 0: +x 1:-x 2:+y 3: -y 
d_safe= p[nx+nx+no:] # [m]

## weights and hovering reference
u_hover=[9.81,0,0]
Q_state=[10.0,5.0,5.0,5.0,1.0,1.0];
Q_hovering=[1.0,1.0,1.0];
Q_u=[1,1,1];
Q_obstacle=[5.0,5.0,5.0,5.0];  # 0: +x 1:-x 2:+y 3: -y

## Constants
g=9.81 # %gravity
mass = 1.1# % of platform, in kilograms
max_thrust = 30# % of platform, in Newtons

a_x=0.1# % Mass normalized drag coefficients
a_y=0.1#
a_z=0.2#

tao_roll=0.5 #; % Time constants in seconds
tao_pitch=0.5 #;

k_roll=1 ##; % Angle gains
k_pitch=1 #;
x0=X

# bounds on input
u_min = [0,-math.pi/10,-math.pi/10]
u_max = [30,math.pi/10,math.pi/10]
umin=np.kron(np.ones((1,N)),u_min)
umin=umin.tolist()
umin=umin[0]
umax=np.kron(np.ones((1,N)),u_max)
umax=umax.tolist()
umax=umax[0]
c=[]

for i in range(N) :
    k = (i ) * nu
    u0=u[k:k+nu]
    p_0=x0[0]#; % position vector
    v=x0[1:4]#; % velocity vector
    roll=x0[4]#; % roll angle
    pitch=x0[5]#; % pitch angle
    T = u0[0]#; % mass normalized thrust
    roll_g = u0[1]#; %desired angles
    pitch_g = u0[2]#;
    #R_x = np.matrix([[1, 0, 0],[ 0, np.cos(roll), -np.sin(roll)],[0, np.sin(roll), np.cos(roll)]])#; %Rotation around x (roll)
    #R_y = np.matrix([[np.cos(pitch), 0, np.sin(pitch)],[ 0, 1, 0],[-np.sin(pitch), 0, np.cos(pitch)]])#; %Rotation around y (pitch)
    #R = R_y*R_x#; % Rotation matrix without rotation around z since coordinates are fixed to the frame
    #x_0 = p_0[0]
    #y_0 = p_0[1]
    #z_0 = p_0[2]
    #v_dot = R*np.matrix([[0],[ 0],[T]]) + np.matrix([[0],[ 0],[-g]]) - np.matrix([[a_x, 0, 0],[ 0,a_y, 0],[ 0, 0, a_z]])*v
    ddx= (cs.sin(pitch)*cs.cos(roll))*T-a_x*x0[1]
    ddy= (-cs.sin(roll))*T-a_y*x0[2]
    ddz=-g+(cs.cos(pitch)*cs.cos(roll))*T-a_z*x0[3]

    roll_dot = (1/tao_roll)*(k_roll*roll_g - roll)

    pitch_dot = (1/tao_pitch)*(k_pitch*pitch_g - pitch)

    #x0[0]+=x0[3]*dt
    #x0[1]+=x0[4]*dt
    x0[0]+=x0[3]*dt
    x0[1]+=ddx*dt
    x0[2]+=ddy*dt
    x0[3]+=ddz*dt
    x0[4]+=roll_dot*dt
    x0[5]+=pitch_dot*dt
    # Obstacle dynamics
    x_obst[0]+=(-x0[1])*dt
    x_obst[1]+=(x0[1])*dt
    x_obst[2]+=(-x0[2])*dt
    x_obst[3]+=(x0[2])*dt
    #x0=[[x],[y],[z],[v_x],[v_y],[v_z],[roll],[pitch]]
    #cost= np.power((x0[0]-x_ref[0]),2)+np.power((x0[1]-x_ref[1]),2)+np.power((x0[2]-x_ref[2]),2)+np.power((x0[3]-x_ref[3]),2)+np.power((x0[4]-x_ref[4]),2)+np.power((x0[5]-x_ref[5]),2) + cost
    #print(Q_state)
    cost+= cs.sumsqr(Q_state*(x0-x_ref))+cs.sumsqr(Q_hovering*(u0-u_hover))
    #A=x0-x_ref
    #B=Q_state*A
    #print(A[5])
    #print(B[5])
    #obstacle avoidance from 2D lidar
    #cost+= Q_obstacle[0]*cs.fmax(d_safe-x_obst[0]+x0[3]*dt,0) #+x
    #cost+= Q_obstacle[1]*cs.fmax(0.0,d_safe-x_obst[1]-x0[3]*dt) #-x
    #cost+= Q_obstacle[2]*cs.fmax(d_safe-x_obst[2]+x0[4]*dt,0) #+y
    #cost+= Q_obstacle[3]*cs.fmax(d_safe-x_obst[3]-x0[4]*dt,0) #-y
    #print(x0[5])
    if i < N - 1:
        cost+=cs.sumsqr(Q_u*(u0-u[(i+1)*nu:(i+1)*nu+nu]))
    # Constraints
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[0]*(d_safe-x_obst[0]-(-x0[1])*dt))) #+x vx
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[1]*(d_safe-(x_obst[1])-x0[1]*dt))) #-x -vx
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[2]*(d_safe-x_obst[2]-x0[2]*dt))) #+y vy
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[3]*(d_safe-(x_obst[3])-(-x0[2])*dt))) #-y -vy

#c = cs.vertcat(cs.fmax(0.0, -u[0] ),
#               cs.fmax(math.pi/2,  u[1] ),
#               cs.fmax(math.pi/2, -u[1] ),
#               cs.fmax(math.pi/2,  u[2] ),
#               cs.fmax(math.pi/2, -u[2] ),
#                 )

bounds= og.constraints.Rectangle(umin,umax)
problem = og.builder.Problem(u, p, cost) \
.with_penalty_constraints(c)  \
.with_constraints(bounds)
          
          
meta = og.config.OptimizerMeta() \
    .with_version("0.0.0") \
    .with_authors(["P. Sopasakis", "E. Fresk"]) \
    .with_licence("CC4.0-By") \
    .with_optimizer_name("the_optimizer") 

build_config = og.config.BuildConfiguration() \
    .with_build_directory("python_build") \
    .with_build_mode("release") \
     .with_build_c_bindings()  \
    .with_rebuild(True)

solver_config = og.config.SolverConfiguration() \
    .with_lfbgs_memory(15) \
    .with_tolerance(1e-5) \
    .with_max_inner_iterations(155)

builder = og.builder.OpEnOptimizerBuilder(problem,
    metadata=meta,
    build_configuration=build_config,
    solver_configuration=solver_config)
builder.enable_tcp_interface()
builder.build()

mng = og.tcp.OptimizerTcpManager('python_build/the_optimizer')
mng.start()

pong = mng.ping() # check if the server is alive
print(pong)
x_init= [0.0,0.0,0.0,0.0,0.0,0.0];
x_ref = [10.0,0.0,0.0,0.0,0.0,0.0];
x_obst = [0.0,3.0,0.0,0.0];
d_safe=[2.0];
x_0=x_init+x_ref + x_obst +d_safe  #concatenate lists in Python
solution = mng.call(x_0) # call the solver over TCP
print(solution)
print('u_0:')
u0=solution[u'solution']
print(u0[0:3])

mng.kill()`

sinashm, 2019-08-17:
`
import casadi.casadi as cs
import opengen as og
import numpy as np
import math

N=10
dt=1.0/10.0
nu=3 #number of inputs
nx=6 #number of states
no=4 #number of obstacles
nd=1 #number of safety distance
cost=0;
u = cs.SX.sym("u", N*nu) # decision variable (nu = 3 T theta_x theta_y)
p = cs.SX.sym("p", nx+nx+no+nd) # states + state_ref (np = 8 p p_dot theta_x theta_y) + obstacles
X= p[0:nx]
x_ref = p[nx:nx+nx]
x_obst= p[nx+nx:nx+nx+no] # 0: +x 1:-x 2:+y 3: -y 
d_safe= p[nx+nx+no:] # [m]

## weights and hovering reference
u_hover=[9.81,0,0]
Q_state=[10.0,5.0,5.0,5.0,1.0,1.0];
Q_hovering=[1.0,1.0,1.0];
Q_u=[1,1,1];
Q_obstacle=[5.0,5.0,5.0,5.0];  # 0: +x 1:-x 2:+y 3: -y

## Constants
g=9.81 # %gravity
mass = 1.1# % of platform, in kilograms
max_thrust = 30# % of platform, in Newtons

a_x=0.1# % Mass normalized drag coefficients
a_y=0.1#
a_z=0.2#

tao_roll=0.5 #; % Time constants in seconds
tao_pitch=0.5 #;

k_roll=1 ##; % Angle gains
k_pitch=1 #;
x0=X

# bounds on input
u_min = [0,-math.pi/10,-math.pi/10]
u_max = [30,math.pi/10,math.pi/10]
umin=np.kron(np.ones((1,N)),u_min)
umin=umin.tolist()
umin=umin[0]
umax=np.kron(np.ones((1,N)),u_max)
umax=umax.tolist()
umax=umax[0]
c=[]

for i in range(N) :
    k = (i ) * nu
    u0=u[k:k+nu]
    p_0=x0[0]#; % position vector
    v=x0[1:4]#; % velocity vector
    roll=x0[4]#; % roll angle
    pitch=x0[5]#; % pitch angle
    T = u0[0]#; % mass normalized thrust
    roll_g = u0[1]#; %desired angles
    pitch_g = u0[2]#;
    #R_x = np.matrix([[1, 0, 0],[ 0, np.cos(roll), -np.sin(roll)],[0, np.sin(roll), np.cos(roll)]])#; %Rotation around x (roll)
    #R_y = np.matrix([[np.cos(pitch), 0, np.sin(pitch)],[ 0, 1, 0],[-np.sin(pitch), 0, np.cos(pitch)]])#; %Rotation around y (pitch)
    #R = R_y*R_x#; % Rotation matrix without rotation around z since coordinates are fixed to the frame
    #x_0 = p_0[0]
    #y_0 = p_0[1]
    #z_0 = p_0[2]
    #v_dot = R*np.matrix([[0],[ 0],[T]]) + np.matrix([[0],[ 0],[-g]]) - np.matrix([[a_x, 0, 0],[ 0,a_y, 0],[ 0, 0, a_z]])*v
    ddx= (cs.sin(pitch)*cs.cos(roll))*T-a_x*x0[1]
    ddy= (-cs.sin(roll))*T-a_y*x0[2]
    ddz=-g+(cs.cos(pitch)*cs.cos(roll))*T-a_z*x0[3]

    roll_dot = (1/tao_roll)*(k_roll*roll_g - roll)

    pitch_dot = (1/tao_pitch)*(k_pitch*pitch_g - pitch)

    #x0[0]+=x0[3]*dt
    #x0[1]+=x0[4]*dt
    x0[0]+=x0[3]*dt
    x0[1]+=ddx*dt
    x0[2]+=ddy*dt
    x0[3]+=ddz*dt
    x0[4]+=roll_dot*dt
    x0[5]+=pitch_dot*dt
    # Obstacle dynamics
    x_obst[0]+=(-x0[1])*dt
    x_obst[1]+=(x0[1])*dt
    x_obst[2]+=(-x0[2])*dt
    x_obst[3]+=(x0[2])*dt
    #x0=[[x],[y],[z],[v_x],[v_y],[v_z],[roll],[pitch]]
    #cost= np.power((x0[0]-x_ref[0]),2)+np.power((x0[1]-x_ref[1]),2)+np.power((x0[2]-x_ref[2]),2)+np.power((x0[3]-x_ref[3]),2)+np.power((x0[4]-x_ref[4]),2)+np.power((x0[5]-x_ref[5]),2) + cost
    #print(Q_state)
    cost+= cs.sumsqr(Q_state*(x0-x_ref))+cs.sumsqr(Q_hovering*(u0-u_hover))
    #A=x0-x_ref
    #B=Q_state*A
    #print(A[5])
    #print(B[5])
    #obstacle avoidance from 2D lidar
    #cost+= Q_obstacle[0]*cs.fmax(d_safe-x_obst[0]+x0[3]*dt,0) #+x
    #cost+= Q_obstacle[1]*cs.fmax(0.0,d_safe-x_obst[1]-x0[3]*dt) #-x
    #cost+= Q_obstacle[2]*cs.fmax(d_safe-x_obst[2]+x0[4]*dt,0) #+y
    #cost+= Q_obstacle[3]*cs.fmax(d_safe-x_obst[3]-x0[4]*dt,0) #-y
    #print(x0[5])
    if i < N - 1:
        cost+=cs.sumsqr(Q_u*(u0-u[(i+1)*nu:(i+1)*nu+nu]))
    # Constraints
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[0]*(d_safe-x_obst[0]-(-x0[1])*dt))) #+x vx
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[1]*(d_safe-(x_obst[1])-x0[1]*dt))) #-x -vx
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[2]*(d_safe-x_obst[2]-x0[2]*dt))) #+y vy
    c= cs.vertcat(c,cs.fmax(0.0,Q_obstacle[3]*(d_safe-(x_obst[3])-(-x0[2])*dt))) #-y -vy

#c = cs.vertcat(cs.fmax(0.0, -u[0] ),
#               cs.fmax(math.pi/2,  u[1] ),
#               cs.fmax(math.pi/2, -u[1] ),
#               cs.fmax(math.pi/2,  u[2] ),
#               cs.fmax(math.pi/2, -u[2] ),
#                 )

bounds= og.constraints.Rectangle(umin,umax)
problem = og.builder.Problem(u, p, cost) \
.with_penalty_constraints(c)  \
.with_constraints(bounds)
          
          
meta = og.config.OptimizerMeta() \
    .with_version("0.0.0") \
    .with_authors(["P. Sopasakis", "E. Fresk"]) \
    .with_licence("CC4.0-By") \
    .with_optimizer_name("the_optimizer") 

build_config = og.config.BuildConfiguration() \
    .with_build_directory("python_build") \
    .with_build_mode("release") \
     .with_build_c_bindings()  \
    .with_rebuild(True)

solver_config = og.config.SolverConfiguration() \
    .with_lfbgs_memory(15) \
    .with_tolerance(1e-5) \
    .with_max_inner_iterations(155)

builder = og.builder.OpEnOptimizerBuilder(problem,
    metadata=meta,
    build_configuration=build_config,
    solver_configuration=solver_config)
builder.enable_tcp_interface()
builder.build()

mng = og.tcp.OptimizerTcpManager('python_build/the_optimizer')
mng.start()

pong = mng.ping() # check if the server is alive
print(pong)
x_init= [0.0,0.0,0.0,0.0,0.0,0.0];
x_ref = [10.0,0.0,0.0,0.0,0.0,0.0];
x_obst = [0.0,3.0,0.0,0.0];
d_safe=[2.0];
x_0=x_init+x_ref + x_obst +d_safe  #concatenate lists in Python
solution = mng.call(x_0) # call the solver over TCP
print(solution)
print('u_0:')
u0=solution[u'solution']
print(u0[0:3])

mng.kill()
`

korken89, 2019-08-18:
I am able to get similar issues if running the solver after freeing it, can this be the issue?

korken89, 2019-08-18:
I have access to the code and had a look around, it seems you are never calling `the_optimizer_new` to create a solver.
Calling `the_optimizer_solve` on this is undefined behavior.

closed 2019-08-18 by sinashm