alphaville/optimization-engine #107: Preconditioning

alphaville, 2019-09-27:
Implemented preconditioning in Python

alphaville, 2021-02-06:
@korken89 This is why this is not so straightforward: we need to compute scaling parameters `w_f`, `w_1` and `w_2` such that the FBE becomes

[image]

Recall that in Rust, we only have access to `F1`, `F2` and `psi`, but not `f` and that `psi` as well as its gradient auto-generated.

The problem is not how we can compute the scaling parameters `w_f`, `w_1` and `w_2`, but how to actually scale `psi` and its gradient.

However, here is a possible easy and, in my opinion, clean solution: Casadi will generate code for the function

[image]

the argument `p` is already there: it is the user-defined parameter; the additional argument `w` will be the vector of scaling parameters. In fact this may or may not be part of `p`.

In other words:

[image]

My proposal would be to use the existing vector `p` to pass `w` because it will be less disruptive.

korken89, 2021-02-06:
Makes sense, I don't remember the interface we use right now, but I understand there is a limit on the interface and we cannot add more parameters? Or is it just simply a simple way to include the new scanning parameters?

alphaville, 2021-02-06:
That's an overview of the various interfaces:

[image]

korken89, 2021-02-06:
Ah okey, it's an issue for the augmented Lagrangian. Does the penalty method have the same issue?

alphaville, 2022-09-17:
@Arun-R-M @smokinmirror To push this forward, let us break it down into small tasks. Every task should correspond to a PR. 

Regarding DevOps:

- create a branch called `dev` out of `master` 
- for every PR, create a branch out of `dev` and after review, merge it into `dev` (not master)
- include unit tests where possible in every PR

Here is a list of tasks:

- [ ] In Python, generate a C file (via CasADi) with the preconditioning parameters (the w's and ρ). Do not change the number of parameters of the solver at this stage
- [ ] In `templates/icasadi/`, in `interface.c`, implement a wrapper in C around the auto-generated files. There are several examples in the same file. You may have to do minor edits in `casadi_memory.h`. You must test this implementation.
- [ ] Move on to `templates/icasadi/icasadi_lib.rs`. This generates the file `icasadi/lib.rs`. There you need to make an interface to allow you to call the C code from Rust. 
- [ ] Modify `templates/optimizer.rs`. This generates the main file `src/lib.rs` in the auto-generated solver. Focus on function `solve`

korken89, 2022-09-17:
🚀 

Happy to see this getting some focus :D

korken89, 2022-09-17:
Also, ping me if you want some feedback :)

alphaville, 2022-09-26:
> Also, ping me if you want some feedback :)

Thank you, Emil. This is almost done now (we're testing it). I'll send you a review request soon.

PR #292 (2022-10-11): Preconditioning
## Main Changes

- Implementation of preconditioning and
- Initial penalty parameter, following the formulas in the book of Birgin and Martínez

To activate the preconditioning use 

```python
solver_cfg = og.config.SolverConfiguration() \
    .with_preconditioning(True)
```

A number of changes have been reviewed in PRs which have been merged into this one. In particular (older to newer),
- #286 
- #287 
- #289
- #290 
- #293
- #295

## Associated Issues

- Closes #106 
- Closes #107

## Releases

- OpEn 0.7.6
- Opengen 0.7.0 [note: it was 0.6.13; according to [SemVer](https:/ [... cut]
closed 2022-10-11 by alphaville