alphaville/optimization-engine #114: Find available port

alphaville, 2019-10-03:
The user can specify a desired port with an option `allow_free_port_discovery`; when activated, OpEn will bind at the next available port. This will allow the user to start multiple OpEn TCP servers without the need to explicitly specify different port. Each port will be stored in the local YAML file of the solver, so the client will know at which port to bind.

When building a TCP server to run remotely, `allow_free_port_discovery` should be most probably disabled.

chutchi2, 2019-11-12:
@alphaville, lets do this

chutchi2, 2019-11-25:
@alphaville, i'm behind on stuff but i'm starting tonight

alphaville, 2019-11-25:
@chutchi2 no worries. If you need to discuss anything, you can reach me on [gitter] tomorrow, or you can drop a comment here.

[gitter]: https://gitter.im/alphaville/optimization-engine?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge

PR #177 (2020-05-11): Release of new versions (opengen v0.5.0, open v0.7.0)
This pull request combines:

- branch `feature/176` which solves issue #176 
- branch `feature/docker-docs` (PR #170)
- branch `dev/ros` (PR #173)

and leads to opengen `v0.5.0` and OpEn Rust library `v0.7.0`. 

### Issues
- closes #114
- closes #149 
- closes #169 
- closes #172 
- closes #175
- closes #176
- closes #178 

### Action list

- [x] update version numbers
- [x] update both CHANGELOG files
- [x] update docs and website
- [x] release OpEn v0.7.0
- [x] release opengen v0.5.0
closed 2020-05-11 by alphaville