alphaville/optimization-engine #120: Not able to build optimizer due to Cargo (editions are unstable)

alejandroastudillo, 2019-10-16:
**Describe the bug**

After updating OpEn to version 0.2.0 (via pip), I am getting the following error:
```
INFO:root:Building optimizer

Caused by:
  editions are unstable

Caused by:
  feature `edition` is required

this Cargo does not support nightly features, but if you
switch to nightly channel you can add
`cargo-features = ["edition"]` to enable this feature
Traceback (most recent call last):
  File "rosenbrock.py", line 28, in <module>
    builder.build()
  File "/home/alejandro/.conda/envs/OpEn/lib/python3.6/site-packages/opengen/builder/optimizer_builder.py", line 530, in build
    self.__build_optimizer()             # build overall project
  File "/home/alejandro/.conda/envs/OpEn/lib/python3.6/site-packages/opengen/builder/optimizer_builder.py", line 394, in __build_optimizer
    raise Exception('Rust build failed')
Exception: Rust build failed

```

**To Reproduce**

Steps to reproduce the behavior:
1. I am just trying to run the rosenbrock example, with some minor modification in SolverConfiguration options:
``` python
import opengen as og
import casadi as cs

# Build parametric optimizer
# ------------------------------------
u = cs.SX.sym("u", 5)
p = cs.SX.sym("p", 2)
phi = og.functions.rosenbrock(u, p)
c = cs.vertcat(1.5 * u[0] - u[1],
               cs.fmax(0.0, u[2] - u[3] + 0.1))
bounds = og.constraints.Ball2(None, 1.5)
problem = og.builder.Problem(u, p, phi) \
    .with_penalty_constraints(c)        \
    .with_constraints(bounds)
build_config = og.config.BuildConfiguration()  \
    .with_build_directory("python_test_build") \
    .with_build_mode("debug")                  \
    .with_tcp_interface_config()
meta = og.config.OptimizerMeta()                   \
    .with_optimizer_name("tcp_enabled_optimizer")
solver_config = og.config.SolverConfiguration() \
    .with_lfbgs_memory(15)                      \
    .with_tolerance(1e-5)                       \
    .with_max_inner_iterations(155)		

builder = og.builder.OpEnOptimizerBuilder(problem, meta,
                                          build_config, solver_config)
builder.build()

# Use TCP server
# ------------------------------------
mng = og.tcp.OptimizerTcpManager('python_test_build/tcp_enabled_optimizer')
mng.start()

mng.ping()
solution = mng.call([1.0, 50.0])
print(solution)

mng.kill()
```

2. See error:
```
python rosenbrock.py 
INFO:root:Initialising builder
INFO:root:Checking user parameters
INFO:root:Creating necessary folders
INFO:root:Copying icasadi interface to target directory
INFO:root:Generating Cargo.toml for target optimizer
INFO:root:Generating icasadi Rust library file
INFO:root:Defining CasADi functions and generating C code
INFO:root:Defining function psi(u, xi, p) and its gradient
INFO:root:Function psi and its gradient (C code)
INFO:root:Defining function F1(u, p)
INFO:root:Mapping F1 (C code)
INFO:root:Defining function F2(u, p)
INFO:root:Mapping F2 (C code)
INFO:root:Generating casadi_memory.h
INFO:root:Generating intercafe.c (C interface)
INFO:root:Generating main code for target optimizer (lib.rs)
INFO:root:Generating build.rs for target optimizer
INFO:root:Generating YAML configuration file
INFO:root:Building optimizer

Caused by:
  editions are unstable

Caused by:
  feature `edition` is required

this Cargo does not support nightly features, but if you
switch to nightly channel you can add
`cargo-features = ["edition"]` to enable this feature
Traceback (most recent call last):
  File "rosenbrock.py", line 28, in <module>
    builder.build()
  File "/home/alejandro/.conda/envs/OpEn/lib/python3.6/site-packages/opengen/builder/optimizer_builder.py", line 530, in build
    self.__build_optimizer()             # build overall project
  File "/home/alejandro/.conda/envs/OpEn/lib/python3.6/site-packages/opengen/builder/optimizer_builder.py", line 394, in __build_optimizer
    raise Exception('Rust build failed')
Exception: Rust build failed

```

**System information:**

 - System/Platform: Intel i7
 - OS: Ubuntu 16.04
 - rustc 1.29.0 (aa3ca1994 2018-09-11)
 - Python 3.6.9
 - rustup show:
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/alejandro/.rustup

stable-x86_64-unknown-linux-gnu (default)
rustc 1.38.0 (625451e37 2019-09-23)

**Additional context**

This error is happening with whatever OCP I try to solve by using OpEn, and didn't happen when using OpEn v.0.1.0.

alphaville, 2019-10-16:
@alejandroastudillo This might be because you have an old version of the Rust compiler. The latest version of `rustc` is `1.38.0`. Can you please run:

```console
rustup update
```

in your terminal and try again?

alejandroastudillo, 2019-10-16:
@alphaville I've already tried with `rustup update` but nothing changed. The problem in the old environment remains, but since I am using conda environments, I just created a new one, installed opengen v.0.2.0 and everything is working fine now.

alphaville, 2019-10-16:
I guess Conda caches certain files and this is what gave rise to this issue. I'm glad it works now. I guess I can close this issue now.

closed 2019-10-16 by alphaville

alejandroastudillo, 2019-10-17:
Thank you very much for your help.