alphaville/optimization-engine #126: Should there be an -O3 in icasadi's build.rs?

alphaville, 2019-10-21:
Not sure whether there should be an optional `-O3` in icasadi's build.rs, that is

```rust
 cc::Build::new()
        .flag_if_supported("-Wall")
        .flag_if_supported("-Wpedantic")
        .flag_if_supported("-Wno-long-long")
        .flag_if_supported("-Wno-unused-parameter")
        .flag_if_supported("-O3")
```

If we compile the Rust code with `--release` does the associated `C` code (auto-generated C files by CasADi) get compiled with appropriate optimization flags?

korken89, 2019-10-21:
It is fine :) cc takes info from the build info: https://docs.rs/cc/1.0.1/cc/struct.Build.html#method.opt_level

alphaville, 2019-10-21:
:+1: Thanks a lot!

closed 2019-10-21 by alphaville