alphaville/optimization-engine #130: import opengen fails

sinashm, 2019-10-22:
**Describe the bug**

In the new version of opengen, when I do import opengen I am getting following error:

>>> import opengen as og
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/opengen/__init__.py", line 2, in <module>
    import opengen.builder
  File "/usr/local/lib/python2.7/dist-packages/opengen/builder/__init__.py", line 1, in <module>
    from .optimizer_builder import *
  File "/usr/local/lib/python2.7/dist-packages/opengen/builder/optimizer_builder.py", line 278
    def __construct_mapping_f1_function(self) -> cs.Function:
                                              ^
SyntaxError: invalid syntax

alphaville, 2019-10-22:
@sinashm it seems that you're using Python 2.7. Can you try again with Python >= 3.5? It would be good to use virtualenv.

sinashm, 2019-10-22:
@alphaville It doesn't give me the same error in the python3. I have some connection refused problem, but I will have a look into it.
ConnectionRefusedError: [Errno 111] Connection refused
Does it mean that Opengen will not support python 2.7 anymore?

alphaville, 2019-10-22:
@sinashm That's correct, we do not support Python 2.7 anymore (see the [installation instructions](https://alphaville.github.io/optimization-engine/docs/installation) of OpEn). It was announced earlier this year that Python 2.7 would retire in September if I remember well. I see now there's an [extension](https://pythonclock.org) till the end of this year, but it will soon be deprecated.

Can you submit your connection issue as a separate one?

closed 2019-10-22 by alphaville