alphaville/optimization-engine #131: Connection Error

sinashm, 2019-10-22:
I am using Python 3.6 for opengen and I am getting following warning and the error. Does this problem due to virtualenv? The TCP IP is:
tcp_config = og.config.TcpServerConfiguration(bind_port=3300)        

WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
WARNING:retry.api:[Errno 111] Connection refused, retrying in 1 seconds...
Traceback (most recent call last):
  File "code_run_ros.py", line 21, in <module>
    mhe.start()
  File "/home/pixy/python-environments/env/lib/python3.6/site-packages/opengen/tcp/optimizer_tcp_manager.py", line 110, in start
    self.ping()
  File "/home/pixy/python-environments/env/lib/python3.6/site-packages/opengen/tcp/optimizer_tcp_manager.py", line 91, in ping
    data = self.__send_receive_data(request)
  File "/home/pixy/python-environments/env/lib/python3.6/site-packages/opengen/tcp/optimizer_tcp_manager.py", line 69, in __send_receive_data
    conn_socket = self.__obtain_socket_connection()
  File "</home/pixy/python-environments/env/lib/python3.6/site-packages/decorator.py:decorator-gen-2>", line 2, in __obtain_socket_connection
  File "/home/pixy/python-environments/env/lib/python3.6/site-packages/retry/api.py", line 74, in retry_decorator
    logger)
  File "/home/pixy/python-environments/env/lib/python3.6/site-packages/retry/api.py", line 33, in __retry_internal
    return f()
  File "/home/pixy/python-environments/env/lib/python3.6/site-packages/opengen/tcp/optimizer_tcp_manager.py", line 65, in __obtain_socket_connection
    s.connect((ip, port))
ConnectionRefusedError: [Errno 111] Connection refused

alphaville, 2019-10-22:
I can't tell - I need to see your code or a minimum working example. You might not have started the TCP server, you might not be calling it properly, or not have set the TCP configuration right.

alphaville, 2019-10-24:
@sinashm any updates on this one?

sinashm, 2019-10-25:
@alphaville I try to build the optimizer again with Python3 but it seems it got stuck.

(env) pixy@pixy:~/catkin_workspaces/catkin_sina/src/nmhe$ python3 NMHE.py
INFO:root:Initialising builder
INFO:root:Checking user parameters
INFO:root:Creating necessary folders
INFO:root:Copying icasadi interface to target directory
INFO:root:Generating Cargo.toml for target optimizer
INFO:root:Generating icasadi Rust library file
INFO:root:Defining CasADi functions and generating C code
INFO:root:Defining function psi(u, xi, p) and its gradient
INFO:root:Function psi and its gradient (C code)
INFO:root:Defining function F1(u, p)
INFO:root:Mapping F1 (C code)
INFO:root:Defining function F2(u, p)
INFO:root:Mapping F2 (C code)
INFO:root:Generating casadi_memory.h
INFO:root:Generating intercafe.c (C interface)
INFO:root:Generating main code for target optimizer (lib.rs)
INFO:root:Generating build.rs for target optimizer
INFO:root:Generating YAML configuration file
INFO:root:Building optimizer

alphaville, 2019-10-25:
It turns out that the problem is that you hadn't included `with_constraints` in the problem formulation, therefore the build fails. What we need to do is to have `NoConstraint()` as the default constraints (instead of `None`) so that users don't have to specify the absence of constraints.

closed 2019-11-26 by alphaville