alphaville/optimization-engine #149: ROS CMakefile template

gmsanchez, 2020-02-17:
Hi! I've been able to migrate a MHE problem that I had running on IPOPT to OpEn. So far so good, I am able to get results in less than a millisecond!

The next step should be to use the generated C code with roscpp in a ROS C++ node. Is there any template or example on how to configure the CMakefile (using cmake or cmake_simple)?

Thanks in advance!

alphaville, 2020-02-17:
Hi @gmsanchez. This is great news.

OpEn can generate C/C++ bindings if you compile using the setting `.with_build_c_bindings()` as discussed in the [documentation](https://alphaville.github.io/optimization-engine/docs/python-c). For that you need to install [clang](https://alphaville.github.io/optimization-engine/docs/installation), which is a requirement of `cbindgen`.

OpEn will generate C bindings as well as an [example C file](https://alphaville.github.io/optimization-engine/docs/python-c#bindings-api). You may then modify this file. Compiling and linking is easy, but I think it would be a very good idea for OpEn to auto-generate a cmake file as well.

alphaville, 2020-02-27:
@gmsanchez  I created a pull request to address this issue. As a first step, I am auto-generating a CMakeLists file that allows you to compile the auto-generated example C file. Once you compile using `with_build_c_bindings()`, you can do:

```bash
cmake .
make run
```

to run the auto-generated example.

This is a first-step to an auto-generated ROS node.

alphaville, 2020-04-20:
**Note:** See [ros_optimization_engine](https://github.com/gmsanchez/ros_optimization_engine)

alphaville, 2020-04-23:
First implementation: https://github.com/alphaville/open_ros

It remains to make Jinja2 templates and support automatic code generation for ROS packages

PR #177 (2020-05-11): Release of new versions (opengen v0.5.0, open v0.7.0)
This pull request combines:

- branch `feature/176` which solves issue #176 
- branch `feature/docker-docs` (PR #170)
- branch `dev/ros` (PR #173)

and leads to opengen `v0.5.0` and OpEn Rust library `v0.7.0`. 

### Issues
- closes #114
- closes #149 
- closes #169 
- closes #172 
- closes #175
- closes #176
- closes #178 

### Action list

- [x] update version numbers
- [x] update both CHANGELOG files
- [x] update docs and website
- [x] release OpEn v0.7.0
- [x] release opengen v0.5.0
PR #173 (2020-05-11): Auto-generated ROS package
### Related Issues
- closes #149 

### Main points
Automatic generation of a ROS package that runs a parametric optimizer.

### Example

Run this Python code:

```py
import opengen as og
import casadi.casadi as cs

u = cs.SX.sym("u", 5)
p = cs.SX.sym("p", 2)
phi = og.functions.rosenbrock(u, p)
c = cs.vertcat(1.5 * u[0] - u[1],
               cs.fmax(0.0, u[2] - u[3] + 0.1))
bounds = og.constraints.Ball2(None, 1.5)
problem = og.builder.Problem(u, p, phi) \
    .with_penalty_constraints(c)        \
    .with_constraints(bounds)
meta = og.config.OptimizerMeta()          [... cut]
closed 2020-05-11 by alphaville