alphaville/optimization-engine #151: Cross compiling for Raspberry Pi

gmsanchez, 2020-02-26:
I want to use OpEn from a ROS C++ node, because of that, I am trying to run OpEn on a Raspberry Pi with its C bindings as a first step.

I am running the current stable KDE Neon distribution, which is based on Ubuntu 18.04
I did the following:

```
rustup target add arm-unknown-linux-gnueabihf
apt install gcc-arm-linux-gnueabihf
```
The ```gcc-arm-linux-gnueabihf``` installs GCC version 7.4.0.

Then I modified the [C/C++ Bindings](https://alphaville.github.io/optimization-engine/docs/python-c) example in the documentation in order to cross compiling for the Raspberry Pi, as mentioned on  [OpEn on Raspberry Pi](https://alphaville.github.io/optimization-engine/blog/2019/03/05/matlab-raspberry). The code I used is attached. After running ```python3 rosenbrock_codegen.py``` I get the errors that can be seen in the attached log file.

Any help would be appreciated. Thanks in advance!

[rosenbrock_codegen.py.txt](https://github.com/alphaville/optimization-engine/files/4258449/rosenbrock_codegen.py.txt)
[rosenbrock_codegen_output.log](https://github.com/alphaville/optimization-engine/files/4258466/rosenbrock_codegen_output.log)

gmsanchez, 2020-02-27:
As a workaround, removed line 49 from the attached code (the one that sets cross compiling with the line ```.with_target_system("arm-unknown-linux-gnueabihf")```) and I did the following:

1. Install rust and clang following the [installation instructions](https://alphaville.github.io/optimization-engine/docs/installation) on the Raspberry Pi.
2. Copy the generated ```the_optimizer``` folder to the Raspberry Pi.
3. Access to the ```the_optimizer``` folder on the Raspberry Pi.
4. Compile the Optimization Engine libraries running
```
cargo clean
cargo build --release
```
5. Compile and run the C bindings following the [C/C++ Bindings](https://alphaville.github.io/optimization-engine/docs/python-c) tutorial
```
gcc optimizer.c -lthe_optimizer -L./target/release -pthread -lm -ldl -o optimizer
LD_LIBRARY_PATH=./target/release ./optimizer
```

gmsanchez, 2020-02-27:
Thanks for the quick reply!

> To cross-compile for a Raspberry Pi you need the target ```arm-unknown-linux-gnueabihf```

As I said in my report, I installed the target architecture for Rust by running the commands
```
rustup target add arm-unknown-linux-gnueabihf
apt install gcc-arm-linux-gnueabihf
```
on my notebook. Is that all I need? Or do I have to do something else?

Then, as the attached code shows, I set up the ```build_config``` as follows
```
build_config = og.config.BuildConfiguration()           \
    .with_rebuild(True)                                 \
    .with_build_mode("release")                         \
    .with_target_system("arm-unknown-linux-gnueabihf")                          \
    .with_build_c_bindings()            # <- The important setting
```
As the documentation says, ```rpi``` is a shorthand for ```arm-unknown-linux-gnueabihf``` so it would be equivalent to use one or another. Right?

After running the attached code, I get the errors attached in the log file and I don't have any cross-compiled executable. Would you mind trying to run the attached code?

> As far as I know, you cannot compile Rust programs directly on Raspberry Pi, although I read somewhere that it is supported on Pi 3.

Ignorance bliss then! Lucky me :smile: 

Thanks in advance!

alphaville, 2020-02-27:
I tried your code and it works on my system, so I suspect it's a matter of configuring your target. 

I see you're getting the message "error: linking with `cc` failed." I suppose this means that your program is trying to link using `cc` instead of the appropriate ARM-specific compiler/linker. 

Can you have a look at https://github.com/japaric/rust-cross#tldr-ubuntu-example? You should make sure that you have updated your `~/.cargo/config` configuration file.

Regarding your other question, yes, `rpi` and `arm-unknown-linux-gnueabihf` are the same thing.

alphaville, 2020-02-27:
@gmsanchez you can try this out by using opengen version [`0.3.2a1`](https://pypi.org/project/opengen/0.3.2a1/). You can install it using 

```console
pip install opengen==0.3.2a1
```

gmsanchez, 2020-02-27:
After adding
```
[target.arm-unknown-linux-gnueabihf]
linker = "arm-linux-gnueabihf-gcc"
```
to my ```~/.cargo/config``` file, the cross-compiling works.

Thanks a lot for the help!

closed 2020-02-27 by gmsanchez