alphaville/optimization-engine #159: Using more than one generated crate in project not possible

michael-p, 2020-04-10:
It seems currently to be impossible to depend on two or more of the crates generated by optimization_engine in a single project because the contained `icasadi` crate creates a name conflict when building the whole project:
```
error: package collision in the lockfile: packages icasadi v0.2.1 (/Users/michaelp/Documents/blimp/mpc/path_follow/icasadi) and icasadi v0.2.1 (/Users/michaelp/Documents/blimp/mpc/position_hold/icasadi) are different, but only one can be written to lockfile unambiguously
```

Also, this prevents having the generated crates be a member of a Cargo workspace (they have to be explicitly excluded otherwise Cargo complains about two crates with same name in a single workspace).

I wasn't able to check yet because it already fails with the above error, but I think the generated `tcp_iface` crates might cause similar problems.

I'm using version 0.6.2, with code generated using the Python interface.

Thanks,
Michael

PS: Besides this minor issue your crate is fantastic!! I was able to get everything up and running in no time, and it works flawlessly! Thank you very much for all your excellent work! :)

alphaville, 2020-04-11:
Hi Michael,

Thank you for reporting this and for your kind words. The crate `icasadi` is intended to be an "internal dependency" that is automatically generated. Since Rust doesn't support currently private dependencies, I think the remedy should be to give `icasadi` a different name for different optimizers. It can be `icasadi_{optimizer_name}`. I will start a new branch to try this - we'll be in touch.

Best regards,
Pantelis

michael-p, 2020-04-12:
I did some more tests, and indeed just renaming the generated `icasadi` crate in its `Cargo.toml` (that is, replacing `name = "icasadi"` with `name = "icasadi_{optimizer_name}"`) and then in the root crates `Cargo.toml` specify the dependency with `icasadi = { package="icasadi_{optimizer_name}", path = "./icasadi/"}` works fine, as you suggested. I just manually changed the generated output as described above and using several optimizer-crates in a single project workspace is fine this way.

alphaville, 2020-04-12:
This should be fixed now. I created an alpha version of `opengen` that you can test. You can install `0.4.1a1` using

```console
pip install opengen==0.4.1a1
```

Can you give it a try and let me know whether it fixes this problem?

closed 2020-04-13 by alphaville

michael-p, 2020-04-14:
Works perfectly, thank you very much!

commit 42b28b52 (2020-04-12): fix #159: tcp interface with project-specific name
PR #163 (2020-04-14): Update webpage
Preview:

[image]

[image]

[image]

[image]

New sidebar that collapses automatically:

[image]

Main changes:

- Overhaul of main page
- Sidebars
- New logo
- Updated docs after fix #159 
- Added metadata for SEO