alphaville/optimization-engine #176: A method to monitor the running solver cost

strimble08, 2020-04-28:
When using the OpEn solver I need to read the total cost at each time step.

A feature that can monitor the running cost, which is calculated using the cost functions provided to the solver. Similar to how solution values are called (solver_status.f1_infeasibility, etc).

alphaville, 2020-05-01:
This is under development in branch `feature/176`

Necessary steps:

- [ ] Implement method `compute_cost_at_solution` in `alm_optimizer.rs`
- [ ] Add field `cost` in `AlmOptimizerStatus`
- [ ] Assign cost value to solution status (in `AlmOptimizer::solve`)
- [ ] Export cost value via TCP interface
- [ ] Read value in TCP client
- [ ] Export value over C/C++ interface (update C bindings)
- [ ] Export value over ROS interface

BjoernLindqvist, 2020-05-07:
This issue is very similar to my own, so I will just comment it here: It would be very valuable and convenient to be able to directly get the cost value of the optimized/converged solution as part of the tcp-message for any stochastic optimization or "trajectory validation" application.

alphaville, 2020-05-07:
@BjoernLindqvist Certainly that will be useful. It's already in the pipeline. I have implemented some necessary things in Rust, so it only remains to expose that information over the various interfaces (see checklist above). I think this should be done in a matter of a few days.

PR #177 (2020-05-11): Release of new versions (opengen v0.5.0, open v0.7.0)
This pull request combines:

- branch `feature/176` which solves issue #176 
- branch `feature/docker-docs` (PR #170)
- branch `dev/ros` (PR #173)

and leads to opengen `v0.5.0` and OpEn Rust library `v0.7.0`. 

### Issues
- closes #114
- closes #149 
- closes #169 
- closes #172 
- closes #175
- closes #176
- closes #178 

### Action list

- [x] update version numbers
- [x] update both CHANGELOG files
- [x] update docs and website
- [x] release OpEn v0.7.0
- [x] release opengen v0.5.0
closed 2020-05-11 by alphaville