alphaville/optimization-engine #195: Python example code does not run

ssokolen, 2020-08-11:
**Describe the bug**

The python example code provided in the GitHub readme (under Code Generation Example) errors out.

`AttributeError: 'SolverConfiguration' object has no attribute 'with_constraints_tolerance'`

opengen was installed with `pip install opengen`

The pip version is likely out of date but if pip is the recommended installation platform, perhaps the first example on GitHub should match the pip version. I should also note that the code from [the website](https://alphaville.github.io/optimization-engine/docs/python-interface) fails with `AttributeError: 'SolverConfiguration' object has no attribute 'with_lfbgs_memory'`

**To Reproduce**

Steps to reproduce the behavior:
1. Copy paste python example code from Code Generation Example
2. Run code with python3
3. See error

**Expected behavior**

First basic example is expected to run without issues.

**System information:**

:warning: Please, provide the following information:

 - System/Platform: Virtual Machine running on PC (i7-6700K CPU)
 - OS: Arch Linux
 - What is the output of `rustup show`?
```
installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu (default)
nightly-x86_64-unknown-linux-gnu

active toolchain
----------------
stable-x86_64-unknown-linux-gnu (default)
rustc 1.42.0 (b8cedc004 2020-03-09)
```
 - What is the output of `rustc -V`?
```
rustc 1.42.0 (b8cedc004 2020-03-09)
```
 - Python/MATLAB version if relevant
```
Python 3.8.5
```

**Additional context**

Output from `pip install opengen`:

```
Successfully installed casadi-3.5.3 decorator-4.4.2 jinja2-2.11.2 opengen-0.5.0 py-1.9.0 pyyaml-5.3.1 retry-0.9.2
```

alphaville, 2020-08-11:
Thank you for reporting this - the documentation is wrong; it should rather be `with_delta_tolerance`. We will fix this as soon as possible.

alphaville, 2020-09-01:
The website is now updated and the typos have been fixed:

- `with_constraints_tolerance` changed to `with_delta_tolerance`
- `with_lfbgs_memory` changed to `with_lbfgs_memory` (should be LBFGS)

closed 2020-09-01 by alphaville