alphaville/optimization-engine #198: TCP Interface to take IP/PORT as cli arguments

davidrusu, 2020-09-08:
**Is your feature request related to a problem? Please describe.**
I'd like to spin up multiple instances of the tcp interface, currently the ip/port is hardcoded in the generated server and this is stopping me from spinning up multiple instances.

**Describe the solution you'd like**
I'd like to have IP/PORT supplied as optional CLI arguments, and the python client would then supply the IP/PORT as arguments when spinning up the TCP server.

alphaville, 2020-09-10:
Very good point. We'll try to address it as soon as possible.

alphaville, 2020-09-10:
An initial implementation is available on #200. The ip and port of the server can be overridden as follows:

```python
mng = og.tcp.OptimizerTcpManager(
    optimizer_path='path/to/optimizer',
    ip='0.0.0.0',
    port=3311)
```

If an IP/port are not provided, the default values will be used. You may of course start multiple instances of the same server listening at different IPs and ports.

If you need to run these servers on different machines, you will most likely need to start them from the terminal using:

```sh
cd tcp_iface_myoptimzer
RUST_LOG=tcp_iface=info cargo run -- --port=8773 --ip=0.0.0.0
```

I will merge it and release a new version after some testing.

davidrusu, 2020-09-10:
💯 Thanks for the speedy response!

alphaville, 2020-09-10:
This is now available in opegen version 0.6.1. You may find documentation [here](https://alphaville.github.io/optimization-engine/docs/python-advanced#tcpip-interface).

closed 2020-09-10 by alphaville