alphaville/optimization-engine #199: Support for other AD libraries

imgeorgiev, 2020-09-09:
Hello

I have been experimenting recently with OpEn and I am very impressed with the performance and optimality of the solver. I would love to use it for an upcoming project but sadly the LGPL license of CasADi prevents me from doing so. Are there plans or discussions about support for other automatic differentiation libraries with non-GPL licenses such as [autodiff](https://github.com/autodiff/autodiff/), [FastAD](https://github.com/JamesYang007/FastAD) or [OpenAD](https://www.mcs.anl.gov/OpenAD/)?

alphaville, 2020-09-10:
Hi Ignat. Support for more AD libraries would be a nice feature to have. One thing you could consider is to use the Rust library of OpEn directly and build interfaces to these libraries.

The way OpEn works with Casadi is that Casadi generates C files, which is accessed via a simpler [C interface](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/templates/icasadi/interface.c) that we developed and finally [interfaced from Rust](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/templates/icasadi/icasadi_lib.rs). 

A similar approach can be followed to interface other libraries as well. The objective will be to implement or auto-generate a C file that implements all methods in [`interface.c`](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/templates/icasadi/interface.c).

It is not in our immediate plans to support this, but we can consider it as a development direction. If you would be willing to work on that, we can arrange a meeting and I can help with the implementation. 

May I also ask why LGPL is not suitable for you?

imgeorgiev, 2021-01-11:
Hi Pantelis

Thank you for the explanation and sorry for the late reply. I am afraid that the project related to my original question has decreased in priority for my company and I do not have the time to contribute to this. However, since I have become more proficient in Rust, I'd actually love to contribute to this project in my own personal time. I would love to see if I can make OpEn work with one of the existing automatic differentiation packages in Rust, turning the library into a standalone Rust solution. I'll get back to you in a month or two when I free up my schedule :)

PS. (L)GPL licences are not suitable for my company and the project I am working on as it requires us to pay royalties to Casadi for using their library.