alphaville/optimization-engine #205: JSON syntax error

Subhankhan41, 2020-09-18:
Hello,
I have run the example_open1.m and getting the following errors:

Warning: Unsuccessful read:  The specified amount of data was not returned within the Timeout period.
'udp' unable to read any data. For more information on possible reasons, see UDP Read Warnings. 
Error using jsondecode
JSON syntax error: expected value but found end of text.

Error in OpEnOptimizer/consume (line 63)
            out = jsondecode(json_response);
Could you please let me know how can I resolve this problem and run the examples correctly? 
Thanks

alphaville, 2020-09-18:
Hi @Subhankhan41. Did you get that error using the MATLAB interface? Can you give some details about the problem you're trying to solve, your operating system, MATLAB version etc?

Subhankhan41, 2020-09-18:
Hello @alphaville I'm ysing Ubuntu 20.04 and MATLAB 2018b. I've tried to run it multiple times but still having the similar error.

alphaville, 2020-09-18:
It seems it is related to the [installation](https://alphaville.github.io/optimization-engine/docs/installation) of clang/libclang. OpEn requires two main dependencies: Rust and clang. You may find more information [here](https://alphaville.github.io/optimization-engine/docs/installation).

Subhankhan41, 2020-09-18:
Yes it was due to clang. Thanks a lot. works perfect.

alphaville, 2020-09-18:
Perfect, I'm glad to hear!

closed 2020-09-18 by alphaville