alphaville/optimization-engine #207: How to add Control Input and state constraints in MATLAB?

Subhankhan41, 2020-09-23:
**Is your feature request related to a problem? Please describe.**
Consider a mobile robot is subject to state and input constraints. How can we set constraints on them? 

**Describe the solution you'd like**
I want to add constraints on input control signal (linear velocity and angular velocity) and states(x, y, theta). Could you please give an example in MATLAB?

**Describe alternatives you've considered**
I have looked in to the closed issue "Min/Max constraint for OpEnConstraints in Matlab #12". However, I'm struggling to add these in the example codes.

alphaville, 2020-09-28:
@Subhankhan41, could you please provide the code you are trying to run so as to be able to reproduce the error?

Subhankhan41, 2020-09-29:
Hello,
Thank you for the reply. Please find the following .txt files( which are .m files): 

1- Example navigation of Kinematic model of mobile robot [example1.txt](https://github.com/alphaville/optimization-engine/files/5296256/example1.txt)
2- constraints [OpEnConstraints.txt](https://github.com/alphaville/optimization-engine/files/5296257/OpEnConstraints.txt)
3- Build [build.txt](https://github.com/alphaville/optimization-engine/files/5296259/build.txt)
For the build/navigation/src/main.rs, I did the modification as follows:
let UMIN : Vec<f64> = vec![-1.0; NU];
let UMAX : Vec<f64> = vec![1.0; NU];
let bounds = Rectangle::new(UMIN, UMAX);
However, it gets updated each time as the file is auto-generated.

Subhankhan41, 2020-10-10:
@alphaville hello again, sorry for asking again but have you resolved it?

alphaville, 2022-02-23:
@Subhankhan41 the issue has been resolved in #265. We will test and merge it soon.

The MATLAB interface is experimental; there are several developments in the the Python interface (additional algorithms, better and more stable implementations, etc) that are missing from the MATLAB interface. If you're fine with this, I would recommend switching to Python for now.

PR #265 (2022-03-04): Support for rectangles in MATLAB
## Main Changes

- Support for rectangles in MATLAB

## Associated Issues

- Resolves #207

## TODOs

- [ ] Documentation 
- [x] All tests must pass
- [ ] Update `CHANGELOG`(s)
- [ ] Update webpage documentation
- [ ] Bump versions (in `CHANGELOG`, `Cargo.toml` and `VERSION`)
commit 830e8e0e (2022-03-04): Resolves #207
closed 2022-03-04 by alphaville