alphaville/optimization-engine #210: Build errors

Mathijssch, 2020-10-06:
**Describe the bug**

In version 0.6.2 of Opengen, the following build error occurs whenever using Cartesian product constraints.

```
error: expected one of `!`, `.`, `::`, `;`, `?`, `{`, `}`, or an operator, found keyword `let`
  --> src/lib.rs:86:9
   |
84 |         bounds
   |               - expected one of 8 possible tokens
85 |     
86 |         let idx_2 = 5;
   |         ^^^ unexpected token

error: aborting due to previous error

error: could not compile `the_optimizer`.
```
It likely has to do with line 200 in `optimization-engine/open-codegen/opengen/templates/optimizer.rs`. 
When I downgrade to version 0.6.0, the problem disappears. 

**To Reproduce**

Steps to reproduce the behavior:

Run the basic example in [this Python module](https://github.com/alphaville/optimization-engine/files/5333805/opentest.txt)

**Expected behavior**

The optimizer to be built without error. 

**System information:**

:warning: Please, provide the following information:

 - System/Platform Intel i7 
 - OS: Ubuntu 20.04
 - What is the output of `rustup show`?
```
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/mathijs/.rustup

stable-x86_64-unknown-linux-gnu (default)
rustc 1.44.1 (c7087fe00 2020-06-17)
```
 - What is the output of `rustc -V`?
```
rustc 1.44.1 (c7087fe00 2020-06-17)
```
 - Python/MATLAB version if relevant
```
>> python
Python 3.7.6 | packaged by conda-forge | (default, Mar 23 2020, 23:03:20) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

alphaville, 2020-10-06:
@Mathijssch Thanks a lot for reporting this. I reproduced the error and I'll try to fix it.

alphaville, 2020-10-06:
I'm pasting this here for reference (from the auto-generated code):

```rust
/// Make constraints U
fn make_constraints() -> impl Constraint {
    // - Cartesian product of constraints:
        let bounds = CartesianProduct::new();
        
        let idx_1 = 2;
        let bounds = bounds.add_constraint(idx_1, NoConstraints::new());
        bounds
    
        let idx_2 = 5;
        let xmin_2 :Option<&[f64]> = Some(&[-1.0,-2.0,-3.0,]);
        let xmax_2:Option<&[f64]> = Some(&[0.0,10.0,-1.0,]);
        let set_2 = Rectangle::new(xmin_2, xmax_2);
        let bounds = bounds.add_constraint(idx_2, set_2);
        bounds
    
    }
```

The function returns `bounds` prematurely.

alphaville, 2020-10-07:
The issue is solved in opengen version [0.6.3](https://pypi.org/project/opengen/0.6.3/) (just released).

closed 2020-10-07 by alphaville