alphaville/optimization-engine #211: TCP remote connection fails if optimizer_path is not given

Maik93, 2020-10-06:
Hi,
I'm trying to connect to a remote TCP server that runs the compiled solver. As exposed here:
https://github.com/alphaville/optimization-engine/blob/d655a88295ea786a8f788f851b589ae2c5c68a7b/open-codegen/opengen/tcp/optimizer_tcp_manager.py#L37-L39

I've written `mng = og.tcp.OptimizerTcpManager(ip="127.0.0.1", port=8333)` but this throws `KeyError: 'build'` on line https://github.com/alphaville/optimization-engine/blob/d655a88295ea786a8f788f851b589ae2c5c68a7b/open-codegen/opengen/tcp/optimizer_tcp_manager.py#L69

It's trivial that only `if optimizer_path is not None` the variable `self.__optimizer_details` is correctly populated (through the call `self.__load_tcp_details()`, I suppose), but if `optimizer_path` is not given, there is simply
https://github.com/alphaville/optimization-engine/blob/d655a88295ea786a8f788f851b589ae2c5c68a7b/open-codegen/opengen/tcp/optimizer_tcp_manager.py#L65

It's clear that it is a dictionary with no `build` key, but I don't know how to fix it, so here I am to write this report ;)

Have a good job and a nice day!

alphaville, 2020-10-06:
Thanks @Maik93 for reporting this. I reproduced the issue and fixed the issue locally. I'll release an updated version of `opengen` and I'll ping you.

alphaville, 2020-10-07:
The issue is solved in opengen version [0.6.3](https://pypi.org/project/opengen/0.6.3/) (just released).

closed 2020-10-07 by alphaville