alphaville/optimization-engine #214: Always building optimizer

chjXu, 2020-10-07:
I learned the Husky example from the `https://alphaville.github.io/optimization-engine/docs/example_navigation_ros_codegen#run-the-example-code-generation-program.`  When I run create_open_solver.py, the program always building optimizer. (Virtual environment is also established according to the website).

python create_open_solver.py 
[INFO] Initialising builder
[INFO] Checking user parameters
[INFO] Creating necessary folders
[INFO] Copying icasadi interface to target directory
[INFO] Generating icasadi's Cargo.toml
[INFO] Generating Cargo.toml for target optimizer
[INFO] Generating icasadi Rust library file
[INFO] Defining CasADi functions and generating C code
[INFO] Defining function psi(u, xi, p) and its gradient
[INFO] Function psi and its gradient (C code)
[INFO] Defining function F1(u, p)
[INFO] Mapping F1 (C code)
[INFO] Defining function F2(u, p)
[INFO] Mapping F2 (C code)
[INFO] Generating casadi_memory.h
[INFO] Generating intercafe.c (C interface)
[INFO] Generating main code for target optimizer (lib.rs)
[INFO] Generating build.rs for target optimizer
[INFO] Generating YAML configuration file
[INFO] Building optimizer

An error will be reported after a few minutes。

[INFO] Initialising builder
[INFO] Checking user parameters
[INFO] Creating necessary folders
[INFO] Copying icasadi interface to target directory
[INFO] Generating icasadi's Cargo.toml
[INFO] Generating Cargo.toml for target optimizer
[INFO] Generating icasadi Rust library file
[INFO] Defining CasADi functions and generating C code
[INFO] Defining function psi(u, xi, p) and its gradient
[INFO] Function psi and its gradient (C code)
[INFO] Defining function F1(u, p)
[INFO] Mapping F1 (C code)
[INFO] Defining function F2(u, p)
[INFO] Mapping F2 (C code)
[INFO] Generating casadi_memory.h
[INFO] Generating intercafe.c (C interface)
[INFO] Generating main code for target optimizer (lib.rs)
[INFO] Generating build.rs for target optimizer
[INFO] Generating YAML configuration file
[INFO] Building optimizer
error: failed to get `libc` as a dependency of package `mpc_controller v0.0.0 (/home/xuchengjun/xcj_ws/open_ros_codegen/nmpc_open/optimization_engine/mpc_controller)`

Caused by:
  failed to load source for dependency `libc`

Caused by:
  Unable to update registry `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to fetch `https://github.com/rust-lang/crates.io-index`

Caused by:
  network failure seems to have happened
  if a proxy or similar is necessary `net.git-fetch-with-cli` may help here
  https://doc.rust-lang.org/cargo/reference/config.html#netgit-fetch-with-cli

Caused by:
  [18] Transferred a partial file; class=Net (12)
Traceback (most recent call last):
  File "create_open_solver.py", line 101, in <module>
    builder.build()
  File "/home/xuchengjun/xcj_ws/open_ros_codegen/venopen/lib/python3.7/site-packages/opengen/builder/optimizer_builder.py", line 572, in build
    self.__build_optimizer()             # build overall project
  File "/home/xuchengjun/xcj_ws/open_ros_codegen/venopen/lib/python3.7/site-packages/opengen/builder/optimizer_builder.py", line 421, in __build_optimizer
    raise Exception('Rust build failed')
Exception: Rust build failed

how should i solve this problem?  Rust, clang, opengen, matplotlib are be installed sucessfully.

alphaville, 2020-10-15:
I see you are getting this error message: `network failure seems to have happened.` Perhaps there was some network error, or crates.io was down while you were trying to run the example. When you run a Rust program that has certain dependencies for the first time, Rust downloads them from crates.io. My guess is that there was some network error the moment you tried. Can you please try again and let me know if the problem persists.

chjXu, 2020-10-15:
> I see you are getting this error message: `network failure seems to have happened.` Perhaps there was some network error, or crates.io was down while you were trying to run the example. When you run a Rust program that has certain dependencies for the first time, Rust downloads them from crates.io. My guess is that there was some network error the moment you tried. Can you please try again and let me know if the problem persists.
> 
> 我看到您得到了这个错误消息: 网络故障似乎已经发生。可能是网络出现了错误，或者在运行示例时 crates.io 出现了故障。当你第一次运行一个有特定依赖项的 Rust 程序时，Rust 会从 crates.io 下载它们。我的猜测是在你尝试的时候出现了一些网络错误。你能再试一次吗? 如果问题仍然存在，请告诉我。

Thanks，when I run it again, there is no problem.