alphaville/optimization-engine #216: PANOC and ALM algorithm

chjXu, 2020-10-15:
I only find panoc and alm algorithm examples by Rust  in example folder. 
Is there some examples by C++ or python?
The best is the example of robotic arm and model predictive control.
Thanks.

alphaville, 2020-10-19:
Good point @chjXu. I will update the documentation with an example on how to build an optimizer for an optimal control problem. This will be useful for MPC and MHE.

alphaville, 2020-10-19:
By the way, you may find examples for Python [here](https://alphaville.github.io/optimization-engine/docs/python-examples).

If you are interested in using OpEn via C or C++, you can use the Python library to generate C/C++ code. It also generates an example file to help you get started.

You will find a lot more on our page at https://alphaville.github.io/optimization-engine/.

closed 2021-11-17 by alphaville