alphaville/optimization-engine #221: roslaunch husky_gazebo husky_empty_world.launch

ding1998, 2021-01-28:
I encountered a problem while running the program of Navigation of ground vehicle using ROS codegen. When I run roslaunch husky_gazebo husky_empty_world.launch, it reports the following error.
... logging to /home/dyh/.ros/log/72e69912-6148-11eb-8456-80fa5b4c850e/roslaunch-dyh-17930.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

resource not found: lms1xx
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_bringup
ROS path [2]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_control
ROS path [3]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_description
ROS path [4]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_desktop
ROS path [5]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_gazebo
ROS path [6]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_msgs
ROS path [7]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_base
ROS path [8]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_navigation
ROS path [9]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_robot
ROS path [10]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_simulator
ROS path [11]=/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_viz
ROS path [12]=/home/dyh/open_ros_codegen/catkin_ws/src/open_nmpc_controller
ROS path [13]=/opt/ros/melodic/share
when processing file: /home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_description/urdf/accessories/sick_lms1xx_mount.urdf.xacro
included from: /home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_description/urdf/husky.urdf.xacro
RLException: while processing /home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_gazebo/launch/spawn_husky.launch:
while processing /home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_control/launch/control.launch:
while processing /home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_description/launch/description.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro','/home/dyh/open_ros_codegen/catkin_ws/src/husky/husky_description/urdf/ husky.urdf.xacro','robot_namespace:=/','laser_enabled:=true','realsense_enabled:=false','urdf_extras:=']] returned with code [2].

Param xml is <param command=``$(find xacro)/xacro'$(find husky_description)/urdf/husky.urdf.xacro' robot_namespace:=$(arg robot_namespace) laser_enabled:=$(arg laser_enabled) realsense_enabled:=$ (arg realsense_enabled) urdf_extras:=$(arg urdf_extras) "name="robot_description"/>
The traceback for the exception was written to the log file

Have you solved the problem，thanks！！

alphaville, 2021-01-28:
Perhaps @gmsanchez will be able to give us a hint since he wrote [this tutorial](https://alphaville.github.io/optimization-engine/docs/example_navigation_ros_codegen).

Could you please share the code of your ROS node (launch file, configuration yaml file, etc) as well as the Python code you used to generate the ROS node?

gmsanchez, 2021-01-28:
@ding1998 you are getting that error because you the package `lms1xx` is missing from your catkin workspace.

I see you are using ROS Melodic, so these instructions might suit you to get a Husky robot and its dependencies up and running

```
cd /path/to/your/catkin_ws/src
git clone https://github.com/husky/husky.git
rosdep install --from-paths /path/to/your/catkin_ws/src --ignore-src
cd /path/to/your/catkin_ws
catkin_make
source /path/to/your/catkin_ws/devel/setup.bash
```

The command `rosdep install --from-paths /path/to/your/catkin_ws/src --ignore-src` checks the missing dependencies for the packages contained in your source directory and tries to resolve them. Try that and tell us how it goes!

ding1998, 2021-01-30:
> @ ding1998，您会收到该错误，因为您的catkin`lms1xx`工作区中缺少该软件包。
> 
> 我看到您正在使用ROS Melodic，因此这些说明可能适合您启动和运行Husky机器人及其依赖项
> 
> ```
> cd /path/to/your/catkin_ws/src
> git clone https://github.com/husky/husky.git
> rosdep install --from-paths /path/to/your/catkin_ws/src --ignore-src
> cd /path/to/your/catkin_ws
> catkin_make
> source /path/to/your/catkin_ws/devel/setup.bash
> ```
> 
> 该命令`rosdep install --from-paths /path/to/your/catkin_ws/src --ignore-src`检查源目录中包含的软件包的缺少依赖项，并尝试解决它们。尝试一下，告诉我们进展如何！

You have provided a very good method, and I have solved my problem according to your method. Thank you very much

ding1998, 2021-01-30:
> gmsanchez也

I have solved my problem according to @gmsanchez's method, thank you very much for your attention

alphaville, 2021-01-30:
The credit goes to @gmsanchez! 

By the way, should we update the documentation to make this clear to the users?

closed 2021-01-30 by alphaville

ding1998, 2021-01-30:
> The credit goes to @gmsanchez!
> 
> By the way, should we update the documentation to make this clear to the users?

I suggest you should do some updates