alphaville/optimization-engine #223: Migrate the repository to GitHub Actions instead of GitHub Actions

korken89, 2021-02-04:
With Travis CI shutting down we need to migrate to GitHub Actions, I have done this for many repositories so I'll fix a PR.

alphaville, 2021-02-04:
I see on their website they've written that "Please be aware travis-ci.org will be shutting down in several weeks, with all accounts migrating to travis-ci.com." Are they actually shutting down?

In any case, 👍 - it will be a good idea to move things to GitHub actions anyway.

korken89, 2021-02-04:
They are shutting down the free service, so you have to pay.
GHA is free :)

alphaville, 2021-02-04:
Let's add this to the README:

[image]

closed 2021-02-05 by alphaville