alphaville/optimization-engine #248: WASM Support

AndrewJSchoen, 2021-11-11:
**Is your feature request related to a problem? Please describe.**
I am working on a project that uses this crate, and we are looking to deploy the project in WebAssembly for use on the web. However, it appears that this library is not currently setup to handle this, and as such it prevents this from being possible as-is. 

**Describe the solution you'd like**
I know of [one fork of this repo](https://github.com/yepw/optimization-engine) that has added this support. Since it would make sense for such a change to be as minimally invasive as possible, it might make sense to configure such implementation as a feature that could be enabled.

**Describe alternatives you've considered**
Currently I am just using the other repo mentioned, but it would be nice to see this in the official repo/crate.

alphaville, 2021-11-11:
Thank you, @AndrewJSchoen for reporting this. Would it be convenient if we supported WebAssembly using a [feature](https://doc.rust-lang.org/cargo/reference/features.html)?

AndrewJSchoen, 2021-11-11:
Yes, that would be fantastic!

alphaville, 2021-11-17:
@AndrewJSchoen  Support for Wasm has been added to OpEn version [0.7.4](https://crates.io/crates/optimization_engine/0.7.4) - just released. You can have a look at [the doucmentation](https://alphaville.github.io/optimization-engine/docs/openrust-features#webassembly-support).

AndrewJSchoen, 2021-11-17:
This looks great! Thanks!

AndrewJSchoen, 2021-12-05:
I was working with the new WASM support, and specified it my cargo.toml as such:
```
optimization_engine = { version = "0.7.4", features = ["wasm"] }
```

, and encountered an error when running it. The only log I was able to retrieve from WebAssembly was 
```
panicked at 'time not implemented on this platform', library/std/src/sys/wasm/../unsupported/time.rs:13:9
```

I tested this again with the old version I was using ([yepw/optimization-engine](https://github.com/yepw/optimization-engine)), and this error did not appear. Maybe something didn't quite make it through to 0.7.4?

alphaville, 2021-12-06:
@AndrewJSchoen thank you for reporting this - let me check.

AndrewJSchoen, 2021-12-06:
Sounds good. Feel free to let me know if you would like me to test/test anything on my end!

AndrewJSchoen, 2022-05-26:
@alphaville Any updates on this? Ideally I would like to switch to the official release.

closed 2023-01-11 by alphaville

alphaville, 2023-01-17:
@AndrewJSchoen  This has been resolved in [OpEn v0.7.7](https://crates.io/crates/optimization_engine) which has been released.

PR #249 (2021-11-17): Support for WASM
Support for Wasm

## Main Changes

- Optional feature `wasm` in `Cargo.toml`
- Using `instant::Instant` instead of `std::Instant` (Wasm-compatible)
- Fixed Rust documentation of `Ball1` 

## Associated Issues

- Closes #248

## TODOs

- [x] Documentation 
- [x] All tests must pass
- [x] Update `CHANGELOG`(s)
- [x] Update webpage documentation
- [x] Bump versions (in `CHANGELOG`, `Cargo.toml` and `VERSION`)

## New Versions

- OpEn v0.7.4
PR #311 (2023-01-11): Fix lingering issue with WASM support
Fixes lingering issue with WASM support

## Main Changes

- Swapped `time::Instant::now()` for `instant::Instant::now()` in panoc optimizer

## Associated Issues

- Closes #248
- Addresses #248

## TODOs

- [ ] Documentation 
- [ ] All tests must pass
- [ ] Update `CHANGELOG`(s)
- [ ] Update webpage documentation
- [ ] Bump versions (in `CHANGELOG`, `Cargo.toml` and `VERSION`)