alphaville/optimization-engine #253: Not able to run open_nav example in Matlab (Windows 10)

sandeep026, 2021-12-01:
**Describe the bug**

Not able to run open_nav example in Matlab (Windows 10) but other examples are working as expected. The following error is shown.

example_open_nav
error: failed to parse manifest at `C:\Users\sunil\Documents\optimization-engine-master\build\navigation\Cargo.toml`

Caused by:
  no targets specified in the manifest
  either src/lib.rs, src/main.rs, a [lib] section, or [[bin]] section must be present
Error using copyfile
The system cannot find the path specified.

Error in OpEnOptimizerBuilder/build>codegen_head (line 147)
copyfile(head_file_path, main_file);

Error in OpEnOptimizerBuilder/build (line 63)
codegen_head(opEnBuilder);

Error in example_open_nav (line 47)
optimizer = builder.build();

rj-mo, 2021-12-02:
We have identified this problem for Windows users and will be bringing out a new release with this fix very soon.

alphaville, 2021-12-03:
@sandeep026 Just a clarification: have you used the method `.with_build_python_bindings()`? Could you share your code and the contents of `C:\Users\sunil\Documents\optimization-engine-master\build\navigation\Cargo.toml`?

alphaville, 2021-12-03:
For reference, here is a [Python notebook](https://colab.research.google.com/drive/1diHoqacIbLG-ojmiABS3JewkHxnZFaf8?usp=sharing) with the navigation example using Python bindings (direct interface).

sandeep026, 2021-12-03:
The code is the example file in the matlab folder example_open_nav.m The contents of the cargo.toml file are listed below.

## cargo.toml
[package]
name = "navigation"
version = "0.1.0"
license = "unspecified"
authors = ["P. Sopasakis <p.sopasakis@gmail.com>", "E. Fresk <emil.fresk@gmail.com>"]
edition = "2018"
publish=false

[dependencies]
optimization_engine = "0.4.0"
icasadi = {path = "./icasadi/"}
serde = { version = "1.0", features = ["derive"] }
serde_json = "1.0"

MuXingchui, 2023-02-17:
> Thanks.
> […](#)

Hi, I meet the same error. Did you fixed the error? If the error has been fixed, could you share your experience?

alphaville, 2023-02-21:
This seems to have been resolved in #318

closed 2023-02-21 by alphaville