alphaville/optimization-engine #256: error: could not compile 'regex-syntax'

achilleas2942, 2021-12-07:
Hello,

I am running the optimizer on a remote cloud server and I am receiving the following error:

error: could not compile 'regex-syntax'

[image]

Do you have a clue? Does it have to do with the server's ports?

alphaville, 2021-12-07:
@achilleas2942 Thank you for reporting this. We've never encountered anything like this before. I don't think it has anything to do with the port. Could you share your Python code or provide a minimum example that replicates the error?

achilleas2942, 2021-12-07:
> @achilleas2942 Thank you for reporting this. We've never encountered anything like this before. I don't think it has anything to do with the port. Could you share your Python code or provide a minimum example that replicates the error?

I ran the same script on several different servers and this error popped up only in one of them..

import opengen as og
import casadi.casadi as cs
import numpy as np

## Problem size

N = 20
dt = 1.0/20
nMAV = 1  #Number of MAVs to be included in the centralized scheme

## Weight matrices
Qx = (8,8, 40, 2, 2, 3, 8, 8)    
#Qx = (4,4, 40, 4, 4, 3, 8, 8)    
P = 2*Qx; #final state weight
Ru = (3, 10, 10) # input weights
Rd = (3, 15, 15) # input rate weights

## Objective function generation
nu = 3; #Number of control inputs per MAV
ns = 8; #Number of states per MAV

np = nMAV*ns + nMAV*ns + nu + nu*nMAV  + 3 + 2
#print(np)
u = cs.SX.sym('u', nu*nMAV*N)
z0 = cs.SX.sym('z0', np)
#print(z0)
x = z0[0:nMAV*ns]
#print(x)
x_ref = z0[nMAV*ns:nMAV*ns + nMAV*ns]
#print(x_ref)
u_ref = z0[nMAV*ns + nMAV*ns:nMAV*ns + nMAV*ns + nu]
#print(u_ref)
u_old = z0[nMAV*ns + nMAV*ns + nu:nMAV*ns + nMAV*ns + nu + nMAV*nu]
f_nmhe = z0[nMAV*ns + nMAV*ns + nu + nu:nMAV*ns + nMAV*ns + nu + nu + 3]
Qx_adapt = z0[nMAV*ns + nMAV*ns + nu + nu + 3:nMAV*ns + nMAV*ns + nu + nu + 3 + 2]

cost = 0
c = 0

for i in range(0, N):
###State Cost 
    for j in range(0,nMAV):
        cost += Qx_adapt[0]*(x[ns*j]-x_ref[ns*j])**2 + Qx_adapt[1]*(x[ns*j+1]-x_ref[ns*j+1])**2 + Qx[2]*(x[ns*j+2]-x_ref[ns*j+2])**2 + Qx[3]*(x[ns*j+3]-x_ref[ns*j+3])**2 + Qx[4]*(x[ns*j+4]-x_ref[ns*j+4])**2 + Qx[5]*(x[ns*j+5]-x_ref[ns*j+5])**2 + Qx[6]*(x[ns*j+6]-x_ref[ns*j+6])**2 + Qx[7]*(x[ns*j+7]-x_ref[ns*j+7])**2

####Input Cost 
    u_n = u[(i*nMAV*nu):(i*nMAV*nu+nu*nMAV)]
    for j in range(0,nMAV):
        cost += Ru[0]*(u_n[nu*j] - u_ref[0])**2 + Ru[1]*(u_n[nu*j+1] - u_ref[1])**2 + Ru[2]*(u_n[nu*j+2] - u_ref[2])**2 #Input weights
        cost += Rd[0]*(u_n[nu*j] - u_old[nu*j])**2 + Rd[1]*(u_n[nu*j+1] - u_old[nu*j+1])**2 + Rd[2]*(u_n[nu*j+2] - u_old[nu*j+2])**2 #Input rate weights

        #Input rate constraints
        c = cs.vertcat(c, cs.fmax(0, u_n[nu*j+1] - u_old[nu*j+1] - 0.05))
        c = cs.vertcat(c, cs.fmax(0, u_old[nu*j+1] - u_n[nu*j+1] - 0.05))
        c = cs.vertcat(c, cs.fmax(0, u_n[nu*j+2] - u_old[nu*j+2] - 0.05))
        c = cs.vertcat(c, cs.fmax(0, u_old[nu*j+2] - u_n[nu*j+2] - 0.05))

        ##Velocity Constraints##
        #c = cs.vertcat(c, cs.fmax(0, x[3] - v_max[0]))
        #c = cs.vertcat(c, cs.fmax(0, -x[3] - v_max[0]))
        #c = cs.vertcat(c, cs.fmax(0, x[4] - v_max[1]))
        #c = cs.vertcat(c, cs.fmax(0, -x[4] - v_max[1]))

    ####State update 
    u_old = u_n
    for j in range(0,nMAV):
        x[ns*j] = x[ns*j] + dt * x[ns*j+3]
        x[ns*j+1] = x[ns*j+1] + dt * x[ns*j+4]
        x[ns*j+2] = x[ns*j+2] + dt * x[ns*j+5]
        x[ns*j+3] = x[ns*j+3] + dt * (cs.sin(x[ns*j+7]) * cs.cos(x[ns*j+6]) * u_n[nu*j] - 0.1 * x[ns*j+3] + f_nmhe[0])
        x[ns*j+4] = x[ns*j+4] + dt * (-cs.sin(x[ns*j+6]) * u_n[nu*j] - 0.1*x[ns*j+4] + f_nmhe[1])
        x[ns*j+5] = x[ns*j+5] + dt * (cs.cos(x[ns*j+7]) * cs.cos(x[ns*j+6]) * u_n[nu*j] - 0.2 * x[ns*j+5] - 9.81 + f_nmhe[2])
        x[ns*j+6] = x[ns*j+6] + dt * ((1 / 0.20) * (u_n[nu*j+1] - x[ns*j+6]))
        x[ns*j+7] = x[ns*j+7] + dt * ((1 / 0.17) * (u_n[nu*j+2] - x[ns*j+7]))
        #print(x[ns*j])
    

umin = [3, -0.25, -0.25] * (N*nMAV)
umax = [15.5, 0.25, 0.25] * (N*nMAV)
bounds = og.constraints.Rectangle(umin, umax)
problem = og.builder.Problem(u, z0, cost).with_penalty_constraints(c) \
.with_constraints(bounds)

tcp_config = og.config.TcpServerConfiguration(bind_port=443)

build_config = og.config.BuildConfiguration()  \
.with_build_directory("MAV") \
.with_build_mode("release") \
.with_tcp_interface_config(tcp_config)
#.with_build_c_bindings()
meta = og.config.OptimizerMeta()       \
.with_optimizer_name("shafter_nmpc_1")
#.with_rebuild(True) 
solver_config = og.config.SolverConfiguration() \
.with_tolerance(1e-5) \
.with_initial_tolerance(1e-5) \
.with_max_duration_micros(40000) \
.with_max_outer_iterations(5) \
.with_penalty_weight_update_factor(5) \
.with_initial_penalty(1000.0) 
builder = og.builder.OpEnOptimizerBuilder(problem, meta,
                                          build_config, solver_config) \
.with_verbosity_level(1)

builder.build()

# Use TCP server
# ------------------------------------
mng = og.tcp.OptimizerTcpManager('MAV/shafter_nmpc_1')
mng.start()
x0 =   [2.0,2.0,1.0,0.0,0.0,0.0,0.0,0.0]*nMAV
xref= [-2.0,1.0,1.0,0.0,0.0,0.0,0.0,0.0]*nMAV
uold =[9.81,0.0,0.0]*nMAV
uref =[9.81,0.0,0.0]
qx_adapt = [5,5]

z0 = x0 + xref + uref + uold + [0,0,0] + qx_adapt
print(z0)
print(len(z0)) ##Length is equal to np
#obsdata = (0.0,0.0,1.0,1.0)
mng.ping()
solution = mng.call(z0, initial_guess=[9.81,0,0.0]*(N*nMAV),buffer_len = 8*4096)
print(solution['solution'])
mng.kill()

achilleas2942, 2021-12-09:
> I got the following error when I tried to call the solver:
> 
> ```
> thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: 
>    Os { code: 13, kind: PermissionDenied, message: "Permission denied" }', src/main.rs:200:88
> ```
> 
> This is because 443 is used for secure communication and I suppose it requires some additional configuration (certificates, etc). I changed the port to `3333` and your code works fine - it prints:
> 
> ```
> [2.0, 2.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, -2.0, 1.0, ...]
> 27
> [9.83681209881683, 0.0501713005271658, -0.05208232314789814, ...]
> ```
> 
> Is it necessary that you use port `443`?

I tried previously several ports but I always receive the same error on a specific server. On other servers (I use port 3047) the script works just fine and print the matrices.

alphaville, 2022-08-08:
@achilleas2942 This seems to me like a network-related issue. I'll close this issue for now, but if you figure out what the issue is, please let me know.

closed 2022-08-08 by alphaville