alphaville/optimization-engine #259: Generated `tcp_iface_...` crate fails to compile

michael-p, 2022-01-17:
Dear all,

the generated code for the TCP-interface fails to compile because the used version of the `clap` crate is not specified, that is, in `tcp_iface_.../Cargo.toml` there is the dependency `clap = "*"` which selects the most recent version of clap, 3.0.7, which the TCP interface crate seems to be not compatible with. Manually changing that line to `clap = "2"` results in the generated project compiling and working just fine.
I'm using the Python code generator, most recent version.

Best,
Michael

rj-mo, 2022-01-21:
Thanks for this information, hotfix coming soon.

ruairi

closed 2022-01-24 by rj-mo

michael-p, 2022-01-26:
Works great, thank you all very much for your work! It's much appreciated!