alphaville/optimization-engine #278: [Error 111]connection refused error

hitdezhazha, 2022-07-25:
I try to call the opengen MPC algorithm to calculate at a constant frequency(for now 25Hz), but every time after servel seconds there comes this error: [Error 111]connection refused.
[image]

alphaville, 2022-08-04:
Could you please provide some additional information? Does the server work and then it fails, or does it never connect?

alphaville, 2022-09-28:
Most likely, this is because you are trying to connect to a server that doesn't bind to the address you have provided. This is a safety precaution: when you design your TCP interface, you need to provide an IP. If the IP you used there is not `192.168.137.1`, the server will not accept requests. You can allow incoming requests at all IPs by using the address `0.0.0.0`. However, without looking at your code I cannot tell what the issue might be. I will close this issue for now, but feel free to comment.

closed 2022-09-28 by alphaville