alphaville/optimization-engine #282: Reduce Git diff noise from name mangling and timestamps

alknemeyer, 2022-08-10:
**Is your feature request related to a problem? Please describe.**
Generating code with `opengen` results in slightly different files each time the generator is run, because of name mangling and timestamps. For example, I might generate a Rust crate with C code and get names like `phi_OjQRvSgxYgGmyFncosSA_incref`, which then change if I run the code again.

Likewise there's a `generated at: ` line which updates each time, which I don't find necessary because the file modification date tells me that.

Both of these make Git diffs a lot more noisy, even if there are no material changes in the generated code:

[image]

**Describe the solution you'd like**
I understand the need for symbol mangling, but it would be nice if users could choose how the name gets mangled, so that it doesn't change on each generation. For example, `phi_$MYORGANISATION_incref`, or maybe allow for the extra bit to come first: `$MYORGANISATION_pi_incref`. In the case of my employer, that would be `phi_kpns_incref` or `kpns_phi_incref`.

I would also like to have the option of turning off the `generated at: XXX` lines.

**Describe alternatives you've considered**
I could fix both of these issues by running a post-generate script -- something that identifies the mangled sections and `generated at: ` lines, and removes them. I will do that if the idea mentioned above isn't implemented. That could also be implemented as a script kept in this repo. It just feels a bit brittle/prone to bugs.

**Additional context**
This is purely about noisy Git diffs. Otherwise, no issues, thank you for creating this software :)

alphaville, 2022-08-10:
I'm addressing this issue in PR #283. I'm removing all those random strings and timestamps. Now the function names will have the form `open_phi_{optimizer_name}`. I am still testing this, but it will be included in version 0.6.13 in a few days.

This may actually have an impact on compilation time the second time one compiles the same optimiser (I need to check). [See also #258]

alknemeyer, 2022-08-11:
Thank you for responding and working on this so incredibly quickly!

alphaville, 2022-09-06:
@alknemeyer Alex, sorry, it took me a while more to merge this. You can upgrade to `0.6.13` and there will be no mangled variable/function names.

closed 2022-09-06 by alphaville

alknemeyer, 2022-09-07:
No problem, thank you!

PR #283 (2022-09-06): Function name mangling in auto-generated code
## Main Changes

- Removing timestamps from auto-generated files
- Removing random strings from function names

## Associated Issues

- Closes #282

## TODOs

- [x] All tests must pass
- [x] Update `CHANGELOG`(s)
- [x] Bump versions (in `CHANGELOG` and `VERSION`)