alphaville/optimization-engine #315: Migrate to Non-Fork Repo

zicklag, 2023-01-29:
To increase discoverability on GitHub it could be good to migrate the repository to a standalone repository so that it will pop up in GitHub searches. This seems to be much more recently updated and was able to find it through the lib.rs listing.

You have to contact GitHub support to have them migrate the repo: https://help.github.com/en/articles/why-are-my-contributions-not-showing-up-on-my-profile#commit-was-made-in-a-fork.

alphaville, 2023-02-14:
@korken89 would it be easier for you to delete the parent repository?

According to [this](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/what-happens-to-forks-when-a-repository-is-deleted-or-changes-visibility#deleting-a-public-repository):

> When you delete a public repository, **one of the existing public forks is chosen to be the new upstream repository**. All other repositories are forked off of this new upstream and subsequent pull requests go to this new upstream repository.

korken89, 2023-02-16:
I can try, lets see what happens

korken89, 2023-02-16:
It is done

closed 2023-02-16 by korken89