alphaville/optimization-engine #316: Print or log out parameter length when compiling the solver

strimble08, 2023-02-14:
**Is your feature request related to a problem? Please describe.**
When creating the solver it would be useful to have a way to return the length of the parameters list. This would allow for faster local debug, or unit testing.

**Describe the solution you'd like**
On compile, the length of the parameter list is returned and can be stored as a variable.

**Describe alternatives you've considered**
The length of the list being sent to the solver can be checked manually at runtime by checking the parameters going in, but there is no way to check the length that the solver expects until there is an error returned.

**Additional context**
Nope.

alphaville, 2023-02-14:
Shane, that's a good point. Just note that the number of parameters is the length of the vector `p`, which is provided by the user. I will improve the error message as well.

alphaville, 2023-02-14:
@strimble08 Now the builder returns this information (in the form of a dictionary). Will this be sufficient?

```
build_config:
  build_c_bindings: false
  build_directory: my_optimizers
  build_mode: debug
  build_python_bindings: true
  open_version: '*'
  rebuild: false
  target_system: null
  tcp_interface_config:
    ip: 127.0.0.1
    port: 8333
meta:
  author:
  - John Smith
  licence: MIT
  name: rosenbrock
  version: 0.0.0
paths:
  target: /Users/username/Documents/Development/OpEn/open-codegen/my_optimizers/rosenbrock
problem:
  dim_constraints_aug_lagrangian: 0
  dim_constraints_penalty: 2
  dim_decision_variables: 5
  dim_parameters: 2
solver_config:
  cbfgs_alpha: null
  cbfgs_epsilon: null
  cbfgs_sy_epsilon: null
  constraints_tolerance: 0.0001
  do_preconditioning: true
  initial_penalty: null
  initial_tolerance: 0.0001
  inner_tolerance_update_factor: 0.1
  lbfgs_memory: 10
  max_duration_micros: 5000000
  max_inner_iterations: 1000
  max_outer_iterations: 30
  penalty_weight_update_factor: 1.5
  sufficient_decrease_coefficient: 0.1
  tolerance: 1.0e-06
```

strimble08, 2023-02-14:
Yes looks great - I can see this info in one dict being useful for more than just the initial feature I suggested.

closed 2023-02-14 by alphaville

alphaville, 2023-02-14:
This is now available in opengen, version [`0.7.1`](https://pypi.org/project/opengen/0.7.1/).

PR #317 (2023-02-14): build() to return dictionary with details
## Main Changes

- Method `build()` returns a dictionary with various information that can assist debuggining 
- Better error reporting in Rust (TCP server)

## Associated Issues

- Closes #316

## TODOs

- [x] Documentation 
- [x] Update `CHANGELOG`(s)
- [x] ~Update webpage documentation~
- [x] Bump versions (in `CHANGELOG`, `Cargo.toml` and `VERSION`)