alphaville/optimization-engine #318: Unable to run the MATLAB examples

MuXingchui, 2023-02-17:
**Describe the bug**

When I run the MATLAB examples, the following error occurred.
```
>>example_open_nav
'cargo' is not recognized as an internal or external command, operable program or batch file.
Error using of copyfile
Could not find location 'D:\appdownload\MATLAB\optimization-engine-master\build\navigation\src\main.rs'.

Error in OpEnOptimizerBuilder/build>codegen_head (line 157)
copyfile(head_file_path, main_file);

Error in OpEnOptimizerBuilder/build (line 157)
codegen_head(opEnBuilder);

Error in example_open_nav (line 157)
optimizer = builder.build();
```
It seems that I don't install `Rust`, but I have installed `Rust`.

**System information:**

 - System/Platform [e.g., Intel i7]
 - OS: [Windows]
 - What is the output of `rustup show`?
```
Default host: x86_64-pc-windows-msvc
rustup home:  C:\Users\MuXingchui\.rustup

stable-x86_64-pc-windows-msvc (default)
rustc 1.67.1 (d5a82bbd2 2023-02-07)
```
 - What is the output of `rustc -V`?
```
rustc 1.67.1 (d5a82bbd2 2023-02-07)
```
 - MATLAB: [R2020b]
----
Looking forward to your reply~ :)

MuXingchui, 2023-02-17:
I restarted the computer and the problem above was fixed. But the new error is occurred.
```
error: failed to parse manifest at `D:\appdownload\MATLAB\optimization-engine-master\build\navigation\Cargo.toml`

Caused by:
  no targets specified in the manifest
  either src/lib.rs, src/main.rs, a [lib] section, or [[bin]] section must be present
Error using of copyfile
Could not find location 'D:\appdownload\MATLAB\optimization-engine-master\build\navigation\src\main.rs'.

Error in OpEnOptimizerBuilder/build>codegen_head (line 157)
copyfile(head_file_path, main_file);

Error in OpEnOptimizerBuilder/build (line 63)
codegen_head(opEnBuilder);

Error in example_open_nav (line 47)
optimizer = builder.build();
```
It looks different from the  #253. Looking forward to your reply~ :)

MuXingchui, 2023-02-18:
I did the method in #254  , it doesn't work. Can someone help a little?

MuXingchui, 2023-02-20:
A strange thing happened, after I installed `clang` through `pacman -S  mingw64/mingw-w64-x86_64-clang`, the program can run. But when I installed `LLVM`, the program doesn't run.

closed 2023-02-20 by MuXingchui

Highlight123, 2024-04-23:
I encountered a similar issue to yours, and I followed your advice to install clang through pacman -S  mingw64/mingw-w64-x86_64-clang, but it still doesn't work. Can you help me?

MuXingchui, 2024-04-24:
> I encountered a similar issue to yours, and I followed your advice to install clang through pacman -S mingw64/mingw-w64-x86_64-clang, but it still doesn't work. Can you help me?

Can you provide a more detailed error message and part of  code?

Highlight123, 2024-04-25:
Thank for your help!
OS: [Windows]
MATLAB: [R2022b]
rustc 1.77.2
clang: 18.1.4; pacman -S  mingw64/mingw-w64-x86_64-clang (I have tried LLVM; the error is same)

Code (https://alphaville.github.io/optimization-engine/docs/matlab-examples)
% parameters
L = 0.5; ts = 0.1;
% Prediction horizon
N = 50; 
% target point and bearing
xref=1; yref=1; thetaref = 0;
% weights
q = 10; qtheta = .1; r = 1;
qN = 10*q; qthetaN = 10*qtheta;
nu = 2; nx = 3;
u = casadi.SX.sym('u', nu*N); 
z0 = casadi.SX.sym('z0', nx);
x = z0(1); y = z0(2); theta = z0(3);
cost = 0;
for t = 1:nu:nu*N
    cost = cost + q*((x-xref)^2 + (y-yref)^2) + qtheta*(theta-thetaref)^2;  
    u_t = u(t:t+1);
    theta_dot = (1/L)*(u_t(2)*cos(theta) - u_t(1)*sin(theta));
    cost = cost + r*(u_t'*u_t);
    x = x + ts * (u_t(1) + L * sin(theta) * theta_dot);
    y = y + ts * (u_t(2) - L * cos(theta) * theta_dot);
    theta = theta + ts * theta_dot;
end
cost = cost + qN*((x-xref)^2 + (y-yref)^2) + qthetaN*(theta-thetaref)^2;
constraints = OpEnConstraints.make_no_constraints();
% build the parametric optimizer
builder = OpEnOptimizerBuilder()...
    .with_problem(u, z0, cost, constraints)...
    .with_build_name('navigation')...
    .with_build_mode('release')...
    .with_fpr_tolerance(1e-4)...
    .with_max_iterations(500);
optimizer = builder.build();

I restarted the computer and still fail to run MATLAB examples
error: failed to parse manifest at `E:\Program Files\MATLAB\R2022b\toolbox\matlab\optimization-engine-master\build\navigation\Cargo.toml`

Caused by:
  no targets specified in the manifest
  either src/lib.rs, src/main.rs, a [lib] section, or [[bin]] section must be present
错误使用 copyfile
找不到位置 'E:\Program
Files\MATLAB\R2022b\toolbox\matlab\optimization-engine-master\build\navigation\src\main.rs'。

出错 OpEnOptimizerBuilder/build>codegen_head (第 157 行)
copyfile(head_file_path, main_file);

出错 OpEnOptimizerBuilder/build (第 63 行)
codegen_head(opEnBuilder);

出错 Free_reference (第 8 行)
optimizer = builder.build();

MuXingchui, 2024-04-26:
> no targets specified in the manifest
> either src/lib.rs, src/main.rs, a [lib] section, or [[bin]] section must be present
> 错误使用 copyfile

`no targets specified in the manifest either src/lib.rs, src/main.rs, a [lib] section, or [[bin]] section must be present` It seems like an error of Rust, you can search for related issues in [Rust Issues](https://github.com/rust-lang/rust/issues), [official community](https://users.rust-lang.org/) or web to see if there are any solutions

MuXingchui, 2024-04-26:
> Thank for your help! OS: [Windows] MATLAB: [R2022b] rustc 1.77.2 clang: 18.1.4; pacman -S mingw64/mingw-w64-x86_64-clang (I have tried LLVM; the error is same)
>
> Code [...]

`no targets specified in the manifest either src/lib.rs, src/main.rs, a [lib] section, or [[bin]] section must be present` It seems like an error of Rust, you can search for related issues in [Rust Issues](https://github.com/rust-lang/rust/issues), [official community](https://users.rust-lang.org/) or web to see if there are any solutions