alphaville/optimization-engine #319: An error "linking with link.exe failed: exit code: 1104"

MuXingchui, 2023-02-27:
I got an error: "linking with `link.exe` failed: exit code: 1104", when I compile the program with `VSCode`.
My program is Python. When I build the optimization solver, I got the error:
```bash
error: linking with `link.exe` failed: exit code: 1104

note: Non-UTF-8 output: LINK : fatal error LNK1104: cannot open file “D:\***\mpc-trajectory-generator-master\mpc_build\mpc_navigation\tcp_iface_mpc_navigation\target\release\build\num-traits-abea29fdb848ac61\build_script_build-abea29fdb848ac61.exe”
```

Besides, my environment:
```bash
Python: 3.7.16
OpenGen: 0.6.4
Rustc: 1.67.1 (d5a82bbd2 2023-02-07)
cargo: 1.67.1 (8ecd4f20a 2023-01-10)
```
I don't know what's wrong, please help me

MuXingchui, 2023-03-06:
Well, I know what's wrong: my file path is to long. [See here for details](https://learn.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk1104?view=msvc-170#paths-that-are-too-long:~:text=Paths%20that%20are%20too%20long). 
But when I build the TCP interface, it has another error:
```
error[E0432]: unresolved import `clap::App`
  --> src\main.rs:16:17
   |
16 | use clap::{Arg, App};
   |                 ^^^ no `App` in the root

error: cannot find macro `value_t` in this scope
   --> src\main.rs:263:16
    |
263 |     let port = value_t!(matches, "port", u32).unwrap_or(BIND_PORT_DEFAULT);
    |   
```
Is there something I've overlooked?
Is there a way to get this working?
Thanks!

alphaville, 2023-03-06:
@MuXingchui can you please provide some more details about this? Can you provide a minimal reproducible example?

MuXingchui, 2023-03-06:
> @MuXingchui can you please provide some more details about this? Can you provide a minimal reproducible example?

Thank you for your reply! Refer to #259 , I know why. Then I updated my `OpEn` to the latest version, my project works good.

closed 2023-03-06 by MuXingchui

MuXingchui, 2023-03-06:
> @MuXingchui can you please provide some more details about this? Can you provide a minimal reproducible example?

Hello @alphaville, I have another question. Does MATLAB interface has the option ".with_aug_lagrangian_constraints"? 
Where can I find all MATLAB interfaces ?

alphaville, 2023-03-06:
No, unfortunately it doesn't. I've put a warning about this [here](https://alphaville.github.io/optimization-engine/docs/matlab-interface). It's not difficult to support this, but I currently don't have the time to work on this.

A workaround could be to design your optimiser in Python and then just use it from MATLAB.

MuXingchui, 2023-03-06:
Understood. But how to call the already built solver by Python in MATLAB?

alphaville, 2023-03-06:
You can use the TCP interface. I mean, you can crate a solver in Python and use `.with_tcp_interface_config()` (see [here](https://alphaville.github.io/optimization-engine/docs/python-interface)). 

The TCP server just expects to receive a JSON-formatted string with the parameter, `p`, and returns another JSON-formatted string with the solution and various statistics (solver time, infeasibilities, etc). The communication protocol is described [here](https://alphaville.github.io/optimization-engine/docs/python-tcp-ip).

I can easily make a little MATLAB function or class that will allow you to call from MATLAB an optimiser that has been created in Python. Would that be of interest?

MuXingchui, 2023-03-06:
> You can use the TCP interface. I mean, you can crate a solver in Python and use `.with_tcp_interface_config()` (see [here](https://alphaville.github.io/optimization-engine/docs/python-interface)). [...]

It is clear! I will try it. Sincere thanks for your guidance! :)