alphaville/optimization-engine #322: roslaunch husky_gazebo husky_empty_world.launch

HatemRaya, 2023-04-14:
Dear all

i am a beginner in Ros and Ubuntu and i was trying to install husky mobile robot for my project and launch it by Ros in gazebo so after successfully installing husky desktop and husky simulator i tried to launch the husky in gazebo by this sequence:

sudo apt-get install ros-noetic-husky-desktop
sudo apt-get install ros-noetic-husky-simulator
roslaunch husky_gazebo husky_empty_world.launch

but, with executing the last code i faced this error
RLException: [husky_empty_world.launch] is neither a launch file in package [husky_gazebo] nor is [husky_gazebo] a launch file name
The traceback for the exception was written to the log file

alphaville, 2023-04-24:
Hatem, I believe @gmsanchez (who wrote this part of the code) will be better suited to help. I will transfer this issue to https://github.com/gmsanchez/open_ros_codegen.

alphaville, 2023-04-24:
@HatemRaya, apparently I cannot transfer the issue to https://github.com/gmsanchez/open_ros_codegen, but I would encourage you to ask there. 

I'm assuming that you are trying to run [this example](https://alphaville.github.io/optimization-engine/docs/example_navigation_ros_codegen).

gmsanchez, 2023-04-28:
Hi, @HatemRaya
I have never installed Husky packages using apt. Have you tried following the instructions on the example that @alphaville mentions? They might be a little outdated, but the main difference is using the Husky code from Clearpath's Github repository.

Please, try that way and let me know how it goes

closed 2023-05-24 by alphaville