alphaville/optimization-engine #348: Can't find extern crate

duartebranco01, 2024-06-07:
**Describe the bug**

Hello, I'm having an error when the Rust code generated by the optimizer builder can't import extern crate:

 --> C:\Users\User.cargo\registry\src\index.crates.io-6f17d22bba15001f\optimization_engine-0.9.0\src\constraints\affine_space.rs:3:1
  |
3 | extern crate modcholesky;
  | ^^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate

error[E0463]: can't find crate for ndarray
 --> C:\Users\User.cargo\registry\src\index.crates.io-6f17d22bba15001f\optimization_engine-0.9.0\src\constraints\affine_space.rs:4:1
  |
4 | extern crate ndarray;
  | ^^^^^^^^^^^^^^^^^^^^^ can't find crate

error[E0433]: failed to resolve: use of undeclared crate or module roots
  --> C:\Users\User.cargo\registry\src\index.crates.io-6f17d22bba15001f\optimization_engine-0.9.0\src\constraints\epigraph_squared_norm.rs:53:32
   |
53 |         let cubic_poly_roots = roots::find_roots_cubic(a3, a2, a1, a0);
   |                                ^^^^^ use of undeclared crate or module roots

These crates exist in the index.crates.io-6f17d22bba15001f folder. 
Do I need to configure some kind of path variable or use specific versions of something? 

**To Reproduce**

Steps to reproduce the behavior:
Run the builder (with the versions of programs indicated bellow):
builder = og.builder.OpEnOptimizerBuilder(problem, meta, build_cfg, solver_cfg)
builder.build()

**Expected behavior**

No problem importing extern crate.

**System information:**

:warning: Please, provide the following information:

 - System/Platform: PC
 - OS: Windows 10
 - rustup version: 1.27.1
 - rustc version:  1.78.0
 - cargo version: 1.78.0
 - Python version:  3.9.13

**Additional context**

None.

alphaville, 2024-06-15:
I've never encountered something like this on Linux or MacOS. The code also works on Google Colab. 

@ruairimoran have you seen anything similar on Windows?

@duartebranco01 a possible explanation can be that you're trying to run the code on a machine without an Internet connection, so cargo fails to download the dependencies.

alphaville, 2024-06-19:
It seems that this error is only likely to happen if the machine on which you're compiling is not connected to the Internet. When we need to use OpEn on a robot or vehicle, I would first compile on my laptop, test the code in simulations, then move it to the robot and compile there with `cargo build`. 

Let me know if this resolves the issue.

alphaville, 2024-08-15:
Fixed in #355

closed 2024-08-15 by alphaville