alphaville/optimization-engine #362: Multiple solvers cause compiling issue on ubuntu (works fine on mac)

wrseif, 2024-10-28:
**Describe the bug**

I am generating 2 solvers and when i try to build my cargo project including both I get this type of error:

```
auto_preconditioning_functions.c:(.text.auto_preconditioning_functions_fmin+0x0): multiple definition of auto_preconditioning_functions_fmin'; /home/gello/control-rs/target/arm-unknown-linux-gnueabihf/release/deps/libicasadi_ik_right-18d9e096eb38edbe.rlib(cd802d741d7fdd7c-auto_preconditioning_functions.o):auto_preconditioning_functions.c:(.text.auto_preconditioning_functions_fmin+0x0): first defined here
```

It seems to only happen for preconditioning functions having the same name.

**To Reproduce**

This is how i am generating the solvers
```
problem = og.builder.Problem(u, p, cost).with_constraints(bounds)
    build_config = og.config.BuildConfiguration() \
        .with_build_directory("solvers_left") \
        .with_build_mode("release") \
        .with_rebuild(True) \
        .with_build_c_bindings() \
        # .with_target_system("rpi")         <-- i tried both with and without this
    meta = og.config.OptimizerMeta() \
        .with_optimizer_name(f"ik_{side}")
    solver_config = og.config.SolverConfiguration() \
        .with_tolerance(1e-7) \
        .with_preconditioning(False)
    builder = og.builder.OpEnOptimizerBuilder(problem,
                                            meta,
                                            build_config,
                                            solver_config)\
                                            .with_generate_not_build_flag(False).with_verbosity_level(0)
```

**System information:**

This issue only occurs on Ubuntu. It happens both on a raspberry pi and on an Orin but it works fine on my mac.

This is `rustup show` for the Orin
```
Default host: aarch64-unknown-linux-gnu 
rustup home:  /home/orin/.rustup
stable-aarch64-unknown-linux-gnu (default)
rustc 1.79.0 (129f3b996 2024-06-10)
```

And this is for Mac
```
Default host: aarch64-apple-darwin
rustup home:  /Users/seif/.rustup
installed toolchains
--------------------
stable-aarch64-apple-darwin
1.80.1-aarch64-apple-darwin (default)
active toolchain
----------------
1.80.1-aarch64-apple-darwin (default)
rustc 1.80.1 (3f5fd8dd4 2024-08-06)
```

I tried both 1.79 and 1.80 rust versions and it still doesn't work.

I also followed what was suggested in #151 but it didn't help

alphaville, 2024-10-29:
@wrseif thank you for reporting this. Just to confirm: are you creating two entirely different solvers? Do they have a different name?

wrseif, 2024-10-29:
yes, i am using different names.

wrseif, 2024-10-31:
@alphaville do you have any update on this? i think it's possibly related to #46 but regarding preconditioning only and not the entire solver

alphaville, 2024-10-31:
@wrseif it is possibly related to #46. I was under the impression that this had been resolved and all function names (except those of `static` functions) were unique. I haven't had time to look into it yet. I'll most like try to resolve this tomorrow.

alphaville, 2024-11-04:
@wrseif, could you please provide a minimal example that will allow me to reproduce the issue?

alphaville, 2024-11-04:
I noticed that in the auto-generated CasADi code, some functions (e.g., `casadi_fmin` and `casadi_sq`) are not declared as `static`. This can cause issues if you're using both **C** files in the same project, although we have created a [relevant unit test](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/test/test_2_solvers.c).

alphaville, 2024-11-07:
The issue has been resolved in opengen v0.9.2.

PR #363 (2024-11-07): Issue with multiple solvers
## Main Changes

- Making some casadi-generated functions `static` (e.g., `casadi_sq` and `casadi_fmin`)

## Associated Issues

- Closes #362

## TODOs

- [x] Documentation 
- [x] All tests must pass
- [x] Update `CHANGELOG`(s)
- [x] ~Update webpage documentation~
- [x] Bump versions (in `CHANGELOG`, `Cargo.toml` and `VERSION`)
closed 2024-11-07 by alphaville