alphaville/optimization-engine #366: Compilation for embedded target

Eryk-Mozdzen, 2024-12-07:
**Describe the bug**

I want to use OPEN on microcontroller, ideally on some Cortex-M4 or Cortex-M33 core. Firstly I'm trying to compile project for something diffrent than my computer (x86_64-unknown-linux-gnu). I saw that in main README there is mentioned Raspberry Pi Pico, so I go for it first.

Using python interface i designed my optimizer and it works perfectly using TCP/IP interface in another python test app. I remove TCP/IP interface generation, add C binding generation option and specify target.

```python
# for python test app
#build = og.config.BuildConfiguration() \
#    .with_build_mode('release')        \
#    .with_tcp_interface_config()

# for RPI
build = og.config.BuildConfiguration()                 \
    .with_target_system('arm-unknown-linux-gnueabihf') \
    .with_build_mode('release')                        \
    .with_build_c_bindings()
```

After compilation I expected static linked library for RPI architecture and no errors, but instead I receive many liner errors from Rust:
```
[INFO] --- Initialising builder: 'open_optimizer'
[INFO] Checking user parameters
[INFO] Creating necessary folders
[INFO] Copying icasadi interface to target directory
[INFO] Generating icasadi's Cargo.toml
[INFO] Generating Cargo.toml for target optimizer
[INFO] Generating icasadi Rust library file
[INFO] Defining CasADi functions and generating C code
[INFO] Defining function psi(u, xi, p) and its gradient
[INFO] Function psi and its gradient (C code)
[INFO] Defining function F1(u, p)
[INFO] Mapping F1 (C code)
[INFO] Defining function F2(u, p)
[INFO] Mapping F2 (C code)
[INFO] Defining preconditioning functions
[INFO] Generating casadi_memory.h
[INFO] Generating intercafe.c (C interface)
[INFO] Generating main code for target optimizer (lib.rs)
[INFO] Generating build.rs for target optimizer
[INFO] Generating YAML configuration file
[INFO] Making CasADi functions static
[INFO] Building optimizer
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/emozdzen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/emozdzen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/emozdzen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/opt/drake/bin:/usr/local/texlive/2024/bin/x86_64-linux:/home/emozdzen/xtensa-lx106-elf/bin:/home/emozdzen/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin:/opt/drake/bin:/opt/ros/iron/bin:/usr/local/texlive/2024/bin/x86_64-linux:/home/emozdzen/xtensa-lx106-elf/bin:/home/emozdzen/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin:/home/emozdzen/.cargo/bin:/home/emozdzen/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" VSLANG="1033" "cc" "-Wl,--version-script=/tmp/rustcfQt8jp/list" "-Wl,--no-undefined-version" "/tmp/rustcfQt8jp/symbols.o" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/open_optimizer.open_optimizer.123b2f22b8cc4874-cgu.0.rcgu.o" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/open_optimizer.open_optimizer.123b2f22b8cc4874-cgu.1.rcgu.o" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/open_optimizer.open_optimizer.123b2f22b8cc4874-cgu.2.rcgu.o" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/open_optimizer.open_optimizer.123b2f22b8cc4874-cgu.3.rcgu.o" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/open_optimizer.8z8dfa2nop71aqrzx965o7rnm.rcgu.o" "-Wl,--as-needed" "-Wl,-Bstatic" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libicasadi_open_optimizer-2dffb4a2eb58bed9.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/liboptimization_engine-14cf928774247997.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/liblbfgs-1cb350dcef0fa0aa.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libroots-05442ddd9bed7419.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libinstant-18a0fbff326f3659.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libmodcholesky-569805b5afd7f8f8.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/librand_xorshift-e50cb334e44b9e5f.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libndarray_rand-c509305c45cc0dec.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/librand_distr-bebae15ee3471821.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/librand-3d19ac018b640b5c.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/librand_chacha-917b2374fc559b6b.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libppv_lite86-a253c61e4325977d.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libzerocopy-7287dc1ccdf11c9f.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libbyteorder-7153e1568e981379.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/librand_core-3e2173f3249dfde2.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libgetrandom-80524e9b366d0ee7.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libcfg_if-4c69056948ad1a49.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libndarray-620b1564021007b8.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libmatrixmultiply-1c174760c3c8ff0d.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libapprox-17be8e6ee9fe333b.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/librawpointer-bace4369fec47e2b.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum-7ea4f6bed919bfb3.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum_iter-f44d2700c2d485d7.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum_rational-d111316d53ba5153.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum_complex-72602c310b8e0fa0.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum_bigint-abe15776e3b31b1e.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum_integer-29218f2ab7241312.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/libnum_traits-f201eafca2007e24.rlib" "/home/emozdzen/repos/uav-tvc-goose/software/simulation/open_optimizer/target/arm-unknown-linux-gnueabihf/release/deps/liblibc-55a48986be3b71e4.rlib" "/home/emozdzen/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/arm-unknown-linux-gnueabihf/lib/lib [... cut ...]
- python 3.10.12
- opengen 0.9.2 (installed via pip)
- cargo 1.82
- rustup 1.27.1
- rustc 1.82

after typing `rustup show` I see:
```
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/emozdzen/.rustup

installed targets for active toolchain
--------------------------------------

arm-unknown-linux-gnueabihf
thumbv7em-none-eabi
thumbv8m.main-none-eabi
x86_64-unknown-linux-gnu

active toolchain
----------------

stable-x86_64-unknown-linux-gnu (default)
rustc 1.82.0 (f6e511eec 2024-10-15)
```

Eryk-Mozdzen, 2025-04-18:
Is there any chance of support on this topic? Even an simple tutorial how to compile for different target?

alphaville, 2025-05-06:
@Eryk-Mozdzen This has proved to be challenging. It used to work just fine. We discovered that you need to set the linker when you compile with cargo. 

Can you try the following: generate code with OpEn, but don't compile it. Then from your terminal, nagivate to the root folder of the generated code (e.g., `my_optimizers/rosenbrock`) and run the following:

```
export CARGO_TARGET_ARM_UNKNOWN_LINUX_GNUEABIHF_LINKER=/usr/bin/arm-linux-gnueabihf-ld; cargo build --target=arm-unknown-linux-gnueabihf
```

This works fine.

Then do the same from within the folder `{root_path}/tcp_iface_rosenbrock`. There we're currently getting an error about `gcc_s` not being found, but we're looking into it.

closed 2025-05-08 by alphaville

alphaville, 2025-05-08:
@Eryk-Mozdzen This issue has now been fixed in [`opengen` v0.9.4](https://pypi.org/project/opengen/) via PR #368. We have added a [test](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/test/test_raspberry_pi.py) in our CI workflow to make sure this works on Ubuntu and MacOS. I also copied the generated binary to a Raspberry Pi and made sure everything works. We have also updated the documentation on the [website](https://alphaville.github.io/optimization-engine/docs/python-advanced#cross-compilation).

PR #368 (2025-05-08): Fix cross compilation issue
This is work in progress

## Main Changes

Firstly, each cargo package has its own `.cargo/config.toml` file with the following contents:

```toml
[target.x86_64-apple-darwin]
rustflags = [
  "-C", "link-arg=-undefined",
  "-C", "link-arg=dynamic_lookup",
]
[target.aarch64-apple-darwin]
rustflags = [
  "-C", "link-arg=-undefined",
  "-C", "link-arg=dynamic_lookup",
]

# This is for cross-compiling to ARM architecture (e.g., Raspberry Pi)
[target.arm-unknown-linux-gnueabihf]
linker="arm-linux-gnueabihf-gcc"
```

Then we can cross-compile with `cargo build --target=ar [... cut]